andrewmcgr/klipper_tmc_autotune #29: Multi Instances

mpx1986, 2023-06-24:
Can you please add a multi instances support to your install script

islonina, 2023-08-01:
Same for me. I use dell for 4 printers and it doesnt see my klipper install. Add this to install bash or please provide manual install

OrByR, 2023-11-22:
Install fails unless klipper.service is found. On systems with more then one instance present klipper.service is replaced with klipper-1.service, klipper-2.service, klipper-3.service etc. Detection should include at least klipper-1.service.

yourbuddydinec, 2024-03-02:
Just ran into this problem too. Hope for an update

mjonuschat, 2025-07-31:
Took a look at this and there are just too many ways to configure multi-instance klipper to make the installer reliably support them. If you run a multi-instance setup the best way to go is to:

1) Clone the klipper_tmc_autotune repository
2) Manually link the `autotune_tmc.py`, `motor_constants.py` and `motor_database.cfg` into the `klippy/extras/` path of your instance(s)
3) Restart the klipper service(s)