andrewmcgr/klipper_tmc_autotune #32: toff cant set to 0

jerem1973, 2023-06-26:
Option 'toff' in section 'autotune_tmc stepper_x' must have minimum of 1

Frix-x, 2023-06-28:
That's true for the original TOFF parameter of the TMC, but in the autotune code, toff = 0 means autocomputation of the correct value as it's done by this line: https://github.com/andrewmcgr/klipper_tmc_autotune/blob/f3d777117c917133628d8bd7bff0bd2396d0af2e/autotune_tmc.py#L248

- If `toff` = 0, then we use an internally autocomputed value that is `int(math.ceil((0.85e-5 * self.fclk - 12)/32))`
- If `toff` is set by the user (higher than 0), then we use it as it's wanted by the user

jerem1973, 2023-06-28:
ok i understand thanks for the reply

CODeRUS, 2023-07-09:
@Frix-x 
>     * If `toff` = 0, then we use an internally autocomputed value that is `int(math.ceil((0.85e-5 * self.fclk - 12)/32))`

but

https://github.com/andrewmcgr/klipper_tmc_autotune/blob/main/autotune_tmc.py#L117

Frix-x, 2023-07-09:
@CODeRUS this is only the "internal" value for the autotuning script.

When we set the driver field there is the proper computation done here: https://github.com/andrewmcgr/klipper_tmc_autotune/blob/2fcd927dbcf35e16b6e3e06677b8b99c657fe20e/autotune_tmc.py#L310

CODeRUS, 2023-07-09:
@Frix-x i se, but doc says:

> When set to 0

But it cannot be manually set to 0, only as default value of parameter

mjonuschat, 2025-09-27:
Closing this out - should there be a valid use case for manually set this value to 0 feel free to open a new issue outlining the use case.