andrewmcgr/klipper_tmc_autotune #44: add chapter in readme to explain how you know that it works?

BelgarionNL, 2023-07-07:
I have no idea if its working I have added everything I think.

#####################################################################
#	A/B Steppers https://github.com/bigtreetech/Manta-M5P/blob/master/README.md
#####################################################################

[stepper_x]
step_pin: PC8
dir_pin: !PC9
enable_pin: !PA15
microsteps: 32
rotation_distance: 40
endstop_pin: tmc2240_stepper_x:virtual_endstop
position_min: -12
position_endstop: 132
position_max: 132
homing_speed: 50
homing_retract_dist: 0
homing_positive_dir: true

[tmc2240 stepper_x]
cs_pin: PD9
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: True
run_current: 0.8 
# stealthchop_threshold: 9999990
diag0_pin: ^!PD3
driver_SGT: 1

[stepper_y]
step_pin: PA10
dir_pin: !PA14
enable_pin: !PA13
microsteps: 32
rotation_distance: 40
endstop_pin: tmc2240_stepper_y:virtual_endstop
position_min: 0
position_endstop: 126
position_max: 126
homing_speed: 50
homing_retract_dist: 0
homing_positive_dir: true

[tmc2240 stepper_y]
cs_pin: PD8
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: True
run_current: 0.8 
# stealthchop_threshold: 9999990
diag0_pin: ^!PD2
driver_SGT: 1

#####################################################################
#	Z Steppers
#####################################################################

##  Front Left
[stepper_z]
step_pin: PC6
dir_pin: PC7
enable_pin: !PA9
rotation_distance: 40
microsteps: 32
endstop_pin: PC3
position_max: 110
position_min: -5
position_endstop: 0.4
homing_speed: 8
second_homing_speed: 3
homing_retract_dist: 5

[tmc2240 stepper_z]
cs_pin: PB10
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: False
run_current: 0.976
# stealthchop_threshold: 999999

##  Rear Center
[stepper_z1]
step_pin: PB12
dir_pin: !PB11
enable_pin: !PA8
rotation_distance: 40
microsteps: 32

[tmc2240 stepper_z1]
cs_pin: PB2
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: False
run_current: 0.976 
# stealthchop_threshold: 999999

##  Front Right
[stepper_z2]
step_pin: PB0
dir_pin: !PB1
enable_pin: !PC4
rotation_distance: 40
microsteps: 32

[tmc2240 stepper_z2]
cs_pin: PA6
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: False
run_current: 0.976
# stealthchop_threshold: 999999

#####################################################################
#	TMC AUTOTUNE
# https://github.com/andrewmcgr/klipper_tmc_autotune
#####################################################################
[autotune_tmc stepper_x]
motor: omc-14hs20-1504s
sgt: 1
tuning_goal: autoswitch

[autotune_tmc stepper_y]
motor: omc-14hs20-1504s
sgt: 1
tuning_goal: autoswitch

[autotune_tmc stepper_z]
motor: omc-17hs19-2004s1

[autotune_tmc stepper_z1]
motor: omc-17hs19-2004s1

[autotune_tmc stepper_z2]
motor: omc-17hs19-2004s1

[autotune_tmc extruder]
motor: ldo-36sth17-1004ahg

#####################################################################
#	Extruder
#
#####################################################################

[extruder]
step_pin: ebb36:PD0
dir_pin: !ebb36:PD1
enable_pin: !ebb36:PD2
rotation_distance: 22.52 # Calibrate rotation distance
gear_ratio: 50:10
microsteps: 32
full_steps_per_rotation: 200
filament_diameter: 1.75
nozzle_diameter: 0.400
heater_pin: ebb36:PB13
sensor_type: MAX31865 
sensor_pin: ebb36:PA4
spi_bus: spi1
rtd_nominal_r: 100
rtd_reference_r: 430
rtd_num_of_wires: 2
max_temp: 290
min_extrude_temp: 150
min_temp: 0
control: pid # PID tune
pid_kp: 30.388
pid_ki: 1.647
pid_kd: 140.164
max_extrude_only_distance: 100.0
max_extrude_cross_section: 5
pressure_advance: 0.025 # calibrate pressure advance
pressure_advance_smooth_time: 0.040

[tmc2209 extruder]
uart_pin: ebb36:PA15
interpolate: false
run_current: 0.3 # set an appropriate current
sense_resistor: 0.110
# stealthchop_threshold: 0

BelgarionNL, 2023-07-07:
`#####################################################################
#	A/B Steppers https://github.com/bigtreetech/Manta-M5P/blob/master/README.md
#####################################################################

[stepper_x]
step_pin: PC8
dir_pin: !PC9
enable_pin: !PA15
microsteps: 32
rotation_distance: 40
endstop_pin: tmc2240_stepper_x:virtual_endstop
position_min: -12
position_endstop: 132
position_max: 132
homing_speed: 50
homing_retract_dist: 0
homing_positive_dir: true

[tmc2240 stepper_x]
cs_pin: PD9
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: True
run_current: 0.8 
# stealthchop_threshold: 9999990
diag0_pin: ^!PD3
driver_SGT: 1

[stepper_y]
step_pin: PA10
dir_pin: !PA14
enable_pin: !PA13
microsteps: 32
rotation_distance: 40
endstop_pin: tmc2240_stepper_y:virtual_endstop
position_min: 0
position_endstop: 126
position_max: 126
homing_speed: 50
homing_retract_dist: 0
homing_positive_dir: true

[tmc2240 stepper_y]
cs_pin: PD8
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: True
run_current: 0.8 
# stealthchop_threshold: 9999990
diag0_pin: ^!PD2
driver_SGT: 1

#####################################################################
#	Z Steppers
#####################################################################

##  Front Left
[stepper_z]
step_pin: PC6
dir_pin: PC7
enable_pin: !PA9
rotation_distance: 40
microsteps: 32
endstop_pin: PC3
position_max: 110
position_min: -5
position_endstop: 0.4
homing_speed: 8
second_homing_speed: 3
homing_retract_dist: 5

[tmc2240 stepper_z]
cs_pin: PB10
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: False
run_current: 0.976
# stealthchop_threshold: 999999

##  Rear Center
[stepper_z1]
step_pin: PB12
dir_pin: !PB11
enable_pin: !PA8
rotation_distance: 40
microsteps: 32

[tmc2240 stepper_z1]
cs_pin: PB2
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: False
run_current: 0.976 
# stealthchop_threshold: 999999

##  Front Right
[stepper_z2]
step_pin: PB0
dir_pin: !PB1
enable_pin: !PC4
rotation_distance: 40
microsteps: 32

[tmc2240 stepper_z2]
cs_pin: PA6
spi_speed: 1000000
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
interpolate: False
run_current: 0.976
# stealthchop_threshold: 999999

#####################################################################
#	TMC AUTOTUNE
# https://github.com/andrewmcgr/klipper_tmc_autotune
#####################################################################
[autotune_tmc stepper_x]
motor: omc-14hs20-1504s
sgt: 1
tuning_goal: autoswitch

[autotune_tmc stepper_y]
motor: omc-14hs20-1504s
sgt: 1
tuning_goal: autoswitch

[autotune_tmc stepper_z]
motor: omc-17hs19-2004s1

[autotune_tmc stepper_z1]
motor: omc-17hs19-2004s1

[autotune_tmc stepper_z2]
motor: omc-17hs19-2004s1

[autotune_tmc extruder]
motor: ldo-36sth17-1004ahg

#####################################################################
#	Extruder
#
#####################################################################

[extruder]
step_pin: ebb36:PD0
dir_pin: !ebb36:PD1
enable_pin: !ebb36:PD2
rotation_distance: 22.52 # Calibrate rotation distance
gear_ratio: 50:10
microsteps: 32
full_steps_per_rotation: 200
filament_diameter: 1.75
nozzle_diameter: 0.400
heater_pin: ebb36:PB13
sensor_type: MAX31865 
sensor_pin: ebb36:PA4
spi_bus: spi1
rtd_nominal_r: 100
rtd_reference_r: 430
rtd_num_of_wires: 2
max_temp: 290
min_extrude_temp: 150
min_temp: 0
control: pid # PID tune
pid_kp: 30.388
pid_ki: 1.647
pid_kd: 140.164
max_extrude_only_distance: 100.0
max_extrude_cross_section: 5
pressure_advance: 0.025 # calibrate pressure advance
pressure_advance_smooth_time: 0.040

[tmc2209 extruder]
uart_pin: ebb36:PA15
interpolate: false
run_current: 0.3 # set an appropriate current
sense_resistor: 0.110
# stealthchop_threshold: 0`

pgroon, 2023-11-25:
For each stepper, you need to run the command `AUTOTUNE_TMC STEPPER=[stepper name here]` in the console of mainsail, fluidd, or whatever you are using. 

I am still not sure if the tuning is persistent or not. You can use the command `DUMP_TMC` ([klipper docs here](https://www.klipper3d.org/G-Codes.html#dump_tmc))  to inspect the current configuration of your steppers.

What I still haven't figured out is whether or not the AUTOTUNE_TMC configuration is in any way persistent, or if I need to run the commands again after each startup.

commit fce24b24 (2025-06-26): docs: How to check if autotune is applied

Closes #44
commit 623b17b8 (2025-09-07): docs: How to check if autotune is applied

Closes #44