andrewmcgr/klipper_tmc_autotune #52: Don`t finds motor definition in printer.cfg

Nights77, 2023-07-22:
The module does not find a new engine in printer.cfg Only after adding the engine to the database it works.

```
Could not find motor definition '[motor_constants ldo-42sth40-1684ac]' required by TMC autotuning. It is not part of the database, please define it in your config!

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

[image]
[image]

zhar0v, 2023-07-25:
You must define motor constant before autotune section

mjonuschat, 2025-06-24:
As stated, motors must be defined before you use them in the autotune sections, closing this issue.

closed 2025-06-24 by mjonuschat