andrewmcgr/klipper_tmc_autotune #53: ldo-42sth40-1684AC to add to database.

kinematicdigit, 2023-07-26:
[motor_constants ldo-42sth40-1684AC]
# Coil resistance, Ohms
resistance: 1.65
# Coil inductance, Henries
inductance: 0.0036
# Holding torque, Nm
holding_torque: 0.45
# Nominal rated current, Amps
max_current: 1.68
# Steps per revolution (1.8deg motors use 200, 0.9deg motors use 400)
steps_per_revolution: 200

Nights77, 2023-07-26:
[motor_constants ldo-42sth40-1684ac]
resistance: 1.65
inductance: 0.0016
holding_torque: 0.50
max_current: 1.68
steps_per_revolution: 200

I have it this way. I did according to the documentation.

any-ao, 2023-08-14:
https://www.onetwo3d.co.uk/wp-content/uploads/2021/04/LDO-42STH40-1684ACVRN-RevA-technical-datasheet.pdf

staticanime, 2023-09-02:
For what it's worth, I have a pull request submitted to add this motor: https://github.com/andrewmcgr/klipper_tmc_autotune/pull/67

mjonuschat, 2025-06-24:
Motor is already in the database as `ldo-42sth40-1684ac`

closed 2025-06-24 by mjonuschat