andrewmcgr/klipper_tmc_autotune #59: Error after adding tmc autotune printer.cfg

aee74, 2023-08-07:
After installing TMC autune, adding update manager to moonraker I've added the tmc autotune section to my printer.cfg
Now I get this error

Internal error during connect: No module named enum

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

```

[tmc2209 extruder]
uart_pin: PC11
tx_pin: PC10
uart_address: 3
run_current: 0.300  #0.350
stealthchop_threshold: 999999
# interpolate: false

#########################
#######AUTO TUNE#########
#########################

[autotune_tmc stepper_x]
motor: creality-42-34

[autotune_tmc stepper_y]
motor: creality-42-34

[autotune_tmc stepper_z]
motor: creality-42-34

[autotune_tmc extruder]
motor: ldo-36sth17-1004ahg
```

Any suggestions on what I need to adjust to remove this error?

aee74, 2023-08-08:
Never mind,  Didn't see an earlier post with the same issue. Solved

closed 2023-08-08 by aee74

thorracin1, 2024-01-27:
how did you fix it?