andrewmcgr/klipper_tmc_autotune #60: Error after update to latest autotune

agdtek, 2023-08-12:
Hi guys,
Im getting this error after doing an update to your script

Internal error during connect: No module named enum

this is the log

Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 121, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/pi/klipper/klippy/extras/autotune_tmc.py", line 2, in <module>
    from enum import Enum
ImportError: No module named enum
webhooks client 1969438632: New connection
webhooks client 1969438632: Client info {'program': 'Moonraker', 'version': 'v0.8.0-86-ga7e154f'}
Unable to issue reset command on MCU 'mcu'
Unable to issue reset command on MCU 'rpi'
Unable to issue reset command on MCU 'sb2040'
webhooks client 1969438632: Disconnected
Restarting printer

agdtek, 2023-08-12:
found solution 
ssh do the following
$ source ~/klippy-env/bin/activate
$ pip install enum
$ deactivate

scheini79, 2023-09-19:
> $ source ~/klippy-env/bin/activate
> $ pip install enum
> $ deactivate

Thanks so much, that's a big relief for me.

Ixian, 2023-12-09:
Thanks! Helped me as well.

mjonuschat, 2025-06-24:
TMC Autotune nowadays requires at least Python 3.3 (preferably 3.9 or newer). As this bug report is for Python 2.7 I'll close it as no longer relevant.