andrewmcgr/klipper_tmc_autotune #63: Incorrect inductance for siboor-35sth52-1204a

gbyleveldt, 2023-08-20:
Hi,

I'm battling to tune this motor and was happy to see it in the DB already. I did, however, find it very weird that such a small motor would have a 35mH inductance, as noted in the DB

`[motor_constants siboor-35sth52-1204a]
#Siboor BOM Voron 0.2 steppers
resistance: 2.3
inductance: 0.035
holding_torque: 0.36
max_current: 1.2
steps_per_revolution: 200`

I've actually measured the value with an inductance meter and can confirm you're missing a zero there. It's a 3.5mH motor, so it should read `inductance: 0.0035`

wight554, 2024-01-09:
Created PR that'll address this https://github.com/andrewmcgr/klipper_tmc_autotune/pull/126

wight554, 2024-01-09:
> Created PR that'll address this #126

@andrewmcgr fyi

mjonuschat, 2025-06-24:
Closing, PR #126 has been merged