andrewmcgr/klipper_tmc_autotune #64: Internal error during ready callback: 'MCU_TMC_uart' object has no attribute 'get_tmc_frequency

LilDominus, 2023-08-27:
Hello, I'm trying to configure auto_autotune on my Ender 6 using SKR mini E3 V3 running v0.0.0-4766-gdgus-reloaded-230526-0-gbaae7f3a-inferred. I encounter this error. One guy on discord said that files need from tmc_autotune is  not yet merge to gdgus-reloaded. Is there anything I can do to resolve this issue? Thank a lot and have a good day!
[image]

Drachenkaetzchen, 2023-09-01:
The error comes from an outdated Klipper version. Update Klipper and try again

Frix-x, 2023-09-01:
Right, the gdgus fork of Klipper that you are using is currently 355 commits behind Klipper main branch and date from more than a year ago. If you want to continue using this fork, you will need to ask his maintainer to update it.

This plugin is designed around the official Klipper repository

mjonuschat, 2025-06-24:
Closing this as it's due to an outdated and unsupported klipper fork.