andrewmcgr/klipper_tmc_autotune #68: Sensorless homing not working

Avasty, 2023-09-05:
Hello, i cannot configure my printer to home with this script. 

Here's my config:
```
[autotune_tmc stepper_x]
motor: moons-ms14hs5p4150
tuning_goal: auto
sg4_thrs: 45
voltage: 24
extra_hysteresis: 1
tbl: 2

[autotune_tmc stepper_y]
motor: moons-ms14hs5p4150
tuning_goal: auto
sg4_thrs: 45
voltage: 24
extra_hysteresis: 1
tbl: 2

[autotune_tmc extruder]
motor: moons-cse14hra1l410a
voltage: 24
```

And there is a dump of stepper_x using DUMP_TMC stepper=stepper_x command 
```
2:33 AM
SG_RESULT: 00000036 sg_result=54
2:33 AM
PWM_AUTO: 00140030 pwm_ofs_auto=48 pwm_grad_auto=20
2:33 AM
PWM_SCALE: 00000030 pwm_scale_sum=48
2:33 AM
PWMCONF: 4f0e0c22 pwm_ofs=34 pwm_grad=12 pwm_freq=2 pwm_autoscale=1 pwm_autograd=1 pwm_reg=15 pwm_lim=4
2:33 AM
DRV_STATUS: 801f00c0 ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) cs_actual=31 stst=1
2:33 AM
CHOPCONF: 230301c3 toff=3 hstrt=4 hend=3 tbl=2 vsense=1 mres=3(32usteps) dedge=1
2:33 AM
MSCURACT: 015d00ba cur_a=186 cur_b=-163
2:33 AM
MSCNT: 00000174 mscnt=372
2:33 AM
TSTEP: 000fffff tstep=1048575
2:33 AM
FACTORY_CONF: 0000000c fclktrim=12
2:33 AM
IOIN: 21000041 enn=1 pdn_uart=1 version=0x21
2:33 AM
OTP_READ: 0000000c otp_fclktrim=12
2:33 AM
IFCNT: 000000a2 ifcnt=162
2:33 AM
GSTAT: 00000000
2:33 AM
GCONF: 000001c4 en_spreadcycle=1 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
2:33 AM
========== Queried registers ==========
2:33 AM
COOLCONF: 00000468 semin=8 seup=3 semax=4
2:33 AM
TCOOLTHRS: 00000125 tcoolthrs=293
2:33 AM
SGTHRS: 000000c8 sgthrs=200
2:33 AM
TPOWERDOWN: 00000014 tpowerdown=20
2:33 AM
TPWMTHRS: 000fffff tpwmthrs=1048575
2:33 AM
IHOLD_IRUN: 000c1f1f ihold=31 irun=31 iholddelay=12
2:33 AM
SLAVECONF: 00000200 senddelay=2
2:33 AM
========== Write-only registers ==========
```

Let me know if I should post something else. :)

xand3r40r93, 2023-09-07:
just disable coolstep for homing

Avasty, 2023-09-07:
Which setting is that?

xand3r40r93, 2023-09-07:
https://github.com/andrewmcgr/klipper_tmc_autotune/issues/12#issuecomment-1684275405

Avasty, 2023-09-07:
And where am I supposed to put that? Into autotune.cfg?

Avasty, 2023-09-08:
Still happening.... I set those commands 
```
set_tmc_field stepper=stepper_y field=tcoolthrs value=1048575
set_tmc_field stepper=stepper_x field=tcoolthrs value=1048575
AUTOTUNE_TMC STEPPER=stepper_y sg4_thrs=75
AUTOTUNE_TMC STEPPER=stepper_x sg4_thrs=75
SAVE_CONFIG
```
I home the printer once all is ok... After that second homing is not working again.

Phil1988, 2023-12-16:
Same here.
Normally my TMC5160 work with sensorless homing at `driver_SGT: 2` and noise level is ok.

My hope was that Autotune can even decrease the sound level and make the motors run cooler.

Reality is, that motors beeing louder (even at idle they wine pretty loud) and sensorless homing doest work any more (sgt set in autotune section, set homing speeds according to description and tried to reduce homing current).

If anyone finds a working solution, just let us know :)