andrewmcgr/klipper_tmc_autotune #75: sorry, but How its works?

EU4AX, 2023-10-15:
Sorry for the noob question, but can somebody explain hot this extension actually works? Installation steps are clear, but what is next, how it should be actually used? Should we run the script before printing to tune driver parameters? Will it be executed and perform adjustments while we are printing. Will it write new parameters to existing cfg or a new one? How can we understand that this extension is actually working (except inviting instrument tuner to listen printer's performance)?
It would be really nice to have few more paragraphs in the read me, explaining how this function works. If it has been described already in some other place, would appreciate if somebody can share link to it.

EricZimmerman, 2023-12-17:
you add it to your config and it adjusts driver settings for your motors. thats it.

viesturz, 2023-12-19:
It relies on the motor database to calculate what are the optimal driver settings. IE based on the resistance, inductively.

In very simple terms the driver does not know how fast the motor reacts to voltage changes and typically yanks it around too hard causing vibrations. This gives the driver reasonable configuration for each motor model that allows it to predict the motor movement more accurately and apply the voltage when it's actually needed.

Bambu does this differently where they try different values and measure the noise level. Each has its own pros and cons.

closed 2025-07-31 by mjonuschat