andrewmcgr/klipper_tmc_autotune #80: Internal error on commands: "HYPERLAPSE" and "UPDATE_DELAYED_GCODE"

Vooks916, 2023-10-22:
When defining the autotune_tmc sections for each motor, I am getting an error related to timelapse? This only occurs when using the autotune on motors z2, z3, and extruder on a V2.4. If I only define autotune for any combination of z0, z1, x, and y, no errors are thrown. Klipper, along with everything else on my printer, is fully up to date.
[image]
[klippy(7).log](https://github.com/andrewmcgr/klipper_tmc_autotune/files/13064705/klippy.7.log)

LongDono, 2023-10-29:
In my Trident config, Klipper will not start if I have all 3 Z motors defined. If I disabled Z1 or Z2, it starts up. 

[image]

Samuel-0-0, 2023-11-18:
Too，it conflict with moonraker-timestamp
https://github.com/mainsail-crew/moonraker-timelapse

closed 2023-11-18 by Vooks916