andrewmcgr/klipper_tmc_autotune #81: Autotune fails when tuning_goal is defined

droans, 2023-10-27:
It appears that `self.tuning_goal` always evaluates to `TuningGoal.AUTO` at [Line 161](https://github.com/andrewmcgr/klipper_tmc_autotune/blob/ad68d53e1e703623ab2a29efa88b12e01ed8d51a/autotune_tmc.py#L161) in `cmd_AUTOTUNE_TMC` even when the tuning goal is set in the config, which causes an error since `self.auto_silent` is not defined. 

It appears the issue is with [Line 153](https://github.com/andrewmcgr/klipper_tmc_autotune/blob/ad68d53e1e703623ab2a29efa88b12e01ed8d51a/autotune_tmc.py#L153) redefining `self.tuning_goal` by grabbing it from the console instead of checking the config.

A safer option would probably be to change this to:

```
...
if 'TUNING_GOAL' in gcmd:
    tgoal = gcmd.get('TUNING_GOAL', None).lower()
else:
    tgoal = self.tuning_goal
...
```

Config:

```
[autotune_tmc stepper_x]
motor: omc-17hs15-1504s
tuning_goal: performance

[autotune_tmc stepper_y]
motor: omc-17hs15-1504s
tuning_goal: performance

[autotune_tmc stepper_z]
motor: omc-17hs15-1504s
tuning_goal: performance

[autotune_tmc extruder]
motor: ldo-36sth20-1004ahg
tuning_goal: silent
```

Log:

```
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/michael/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/michael/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/michael/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/michael/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/michael/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/michael/klipper/klippy/gcode.py", line 144, in <lambda>
    handler = lambda gcmd: self._cmd_mux(cmd, gcmd)
  File "/home/michael/klipper/klippy/gcode.py", line 302, in _cmd_mux
    values[key_param](gcmd)
  File "/home/michael/klipper/klippy/extras/autotune_tmc.py", line 161, in cmd_AUTOTUNE_TMC
    self.tuning_goal = TuningGoal.SILENT if self.auto_silent else TuningGoal.PERFORMANCE
AttributeError: 'AutotuneTMC' object has no attribute 'auto_silent'
```

zapl, 2023-11-08:
There's an issue with auto_silent being undefined when using `AUTOTUNE_TMC TUNING_GOAL=auto` when the original config was not auto.

The fix here would be to modify ~ line 110 to eg
```
        if self.tuning_goal == TuningGoal.AUTO:
            # Very small motors may not run in silent mode.
            self.auto_silent = not self.name in AUTO_PERFORMANCE_MOTORS and motor.T > 0.3
            self.tuning_goal = TuningGoal.SILENT if self.auto_silent else TuningGoal.PERFORMANCE
        else:
            self.auto_silent = False # ! needs to be defined
``` 
or to merge the boolean conditions into an unconditional assignment. That would allow the code in
```
self.tuning_goal = TuningGoal.SILENT if self.auto_silent else TuningGoal.PERFORMANCE
```
to work because auto_silent is now always defined

bennierex, 2023-11-19:
Same happens when you have specified a tuning goal in the config and you call the AUTOTUNE_TMC macro with any other parameters.

mjonuschat, 2025-06-25:
Closing this auto since this bug has been resolved a while ago, verified working with the following config:

```
[autotune_tmc stepper_x]
motor = omc-17hs19-2504s-h
tuning_goal = performance
voltage = 48
sgt = 1
```

closed 2025-06-25 by mjonuschat