andrewmcgr/klipper_tmc_autotune #84: What sections can I remove after switching to autotune?

emrebag, 2023-11-01:
Just finished configuring autotune, and noticed that my steppers returned to humming on idle - like they did before setting my stealthchop_threshold to 1 from 0.
I am guessing this klipper is no longer reading this setting from the printer.cfg.
What other lines are no longer relevant when autotune is active? 
ex. can I remove the run_current lines from my config as well?

Also, is there a way to get the minimum stealthchop_threshold back to 1 with autotune active so my steppers stop humming so much on idle?

mjonuschat, 2025-06-25:
Does the [Adjusting existing configuration](https://github.com/andrewmcgr/klipper_tmc_autotune?tab=readme-ov-file#adjusting-existing-configuration) section in the readme help? If not, how can we make it clearer?

mjonuschat, 2025-06-25:
There currently is no way to override the stealthchop treshold as that is set as part of the tuning goal.

mjonuschat, 2025-07-31:
I’m closing this old issue.

closed 2025-07-31 by mjonuschat