andrewmcgr/klipper_tmc_autotune #89: TMC error using fysetc-42HC40-204A-300N84

WaschBaer0395, 2023-11-14:
following error happens when i try to use autotune with fysetc-42HC40-204A-300N84:

TMC 'stepper_z' reports error: DRV_STATUS: 40150010 s2vsa=1(ShortToSupply_A!) cs_actual=21 stealth=1
the motors and drivers work fine if i comment out the whole auto tuning section for the Z, Z1, and Z2 drivers

```
[autotune_tmc stepper_z]
motor: fysetc-42HC40-204A-300N84
#tuning_goal: silent
##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z]
uart_pin: PA15
uart_address: 0
#interpolate: True
run_current: 1.2
hold_current: 1.0
sense_resistor: 0.110
#stealthchop_threshold: 0

[autotune_tmc stepper_z1]
motor: fysetc-42HC40-204A-300N84
#tuning_goal: silent
##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z1]
uart_pin: PC14
#interpolate: True
run_current: 1.2
hold_current: 1.0
sense_resistor: 0.110
#stealthchop_threshold: 0

[autotune_tmc stepper_z2]
motor: fysetc-42HC40-204A-300N84
#tuning_goal: silent
##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z2]
uart_pin: PC15
#interpolate: true
run_current: 1.2
hold_current: 1.0
sense_resistor: 0.110
#stealthchop_threshold: 0
```

bra1nDeaD, 2024-04-01:
I have something very similar. Motors also work when commenting out autotune config, but when using autotune I get these errors the moment it tries to home (different instances):

!! TMC 'stepper_z' reports error: DRV_STATUS: 001f1000 s2vsa=1(ShortToSupply_A!) csactual=31
!! TMC 'stepper_z' reports error: DRV_STATUS: 401f2000 s2vsb=1(ShortToSupply_B!) csactual=31 olb=1(OpenLoad_B!)

Mine are the original creality z motors (creality-42-34).

Wondering if you've found a resolution for this, or if someone else knows what's wrong?

mjonuschat, 2025-06-25:
Catching up on old bug reports, is this still happening on the latest autotune version?

mjonuschat, 2025-07-31:
I’m closing this old issue. If it’s still happening, please open a new one.

closed 2025-07-31 by mjonuschat