andrewmcgr/klipper_tmc_autotune #97: QOL Motor name forcer convert to lowercase

Obscured84, 2023-12-08:
Not an issue, Just a bit of improvement (maybe I'm wrong)
forcing lowercase in motor name

Makes it easier to copy & paste motor names from datasheets

# AutotuneTMC config parameters
        self.motor = config.get('motor')
        self.motor_name = "motor_constants " + self.motor
        try:
            motor = self.printer.lookup_object(self.motor_name.lower())

Obscured84, 2023-12-08:
created pull request [QOL- forcing lowercase when reading motorname from config](https://github.com/andrewmcgr/klipper_tmc_autotune/pull/98)

mjonuschat, 2025-06-24:
Closing this as it was merged in #98 and subsequently reverted due to breaking lookups in #120

closed 2025-06-24 by mjonuschat