andrewmcgr/klipper_tmc_autotune #116: Nothing happens?

michaelmbms, 2024-01-03:
Hello installed your extension without any errors.
And did the necessary changes in printer.cfg.

Gave the command "AUTOTUNE_TMC STEPPER=stepper_x" but nothings happens.
No error, no message its done, and there is no change in any .cfg.

Did i do something wrong?

michaelmbms, 2024-01-03:
Already found out how it works.

closed 2024-01-03 by michaelmbms

locki-cz, 2024-01-22:
Just check klipper logs, it doesnt show anything in console ...

my settings for Z motors on Voron 2.4:

[autotune_tmc stepper_z]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

[autotune_tmc stepper_z1]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

[autotune_tmc stepper_z2]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

[autotune_tmc stepper_z3]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

**then in klippy log:**
autotune_tmc set stepper_z hstrt=4
autotune_tmc set stepper_z hend=3
autotune_tmc set stepper_z pwm_freq=2
autotune_tmc set stepper_z sgthrs=10
autotune_tmc using max PWM speed 189.394382
autotune_tmc set stepper_z pwm_autoscale=True
autotune_tmc set stepper_z pwm_autograd=True
autotune_tmc set stepper_z pwm_grad=16
autotune_tmc set stepper_z pwm_ofs=17
autotune_tmc set stepper_z pwm_reg=15
autotune_tmc set stepper_z pwm_lim=4
autotune_tmc set stepper_z tpwmthrs=0
autotune_tmc set stepper_z en_spreadcycle=False
autotune_tmc set stepper_z tbl=1
autotune_tmc set stepper_z toff=3
autotune_tmc set stepper_z tcoolthrs=59(32.0)
autotune_tmc set stepper_z semin=8
autotune_tmc set stepper_z semax=4
autotune_tmc set stepper_z seup=3
autotune_tmc set stepper_z sedn=0
autotune_tmc set stepper_z seimin=0
autotune_tmc set stepper_z iholddelay=12
autotune_tmc set stepper_z multistep_filt=True

iz3man, 2024-01-22:
Thank you! So basically you need to add (at least) a tuning goal in the cfg to make it do something.
That's what I did now (remote, as I'm in the office, and I planned to test in the evening).

But then: Why is this not stated in the README? Is it only you and me that confused?!
First thing I always look for are config examples so I can adopt those. :)

locki-cz, 2024-01-22:
> Thank you! So basically you need to add (at least) a tuning goal in the cfg to make it do something. That's what I did now (remote, as I'm in the office, and I planned to test in the evening).
> 
> But then: Why is this not stated in the README? Is it only michaelmbms and me that were confused?! First thing I always look for are config examples so I can adopt those. :)

I am not confused, using this for very long time ... almost from begining ...

iz3man, 2024-01-22:
> I am not confused, using this for very long time ... almost from begining ...

I found some old README file, and there it was cleary stated how it is supposed to look like. No idea why it this "autotune" line was removed from it.

iz3man, 2024-01-22:
So now I at least get some ouput in klippy.log. But what to do with it? Do I manually add it to some printer.cfg? Or is this just for information? What to do with this data?!

`AUTOTUNE_TMC stepper_x
autotune_tmc set stepper_x hstrt=0
autotune_tmc set stepper_x hend=3
autotune_tmc using max PWM speed 211.903439
autotune_tmc set stepper_x tbl=1
autotune_tmc set stepper_x toff=3
autotune_tmc set stepper_x sgt=1
autotune_tmc set stepper_x semin=8
autotune_tmc set stepper_x semax=4
autotune_tmc set stepper_x seup=3
autotune_tmc set stepper_x sedn=0
autotune_tmc set stepper_x seimin=0
autotune_tmc set stepper_x sfilt=0`

iz3man, 2024-01-22:
I guess now I got it. You set the config in tmc_autotune.cfg with some tuning_goal=xxx and then this is run AUTOMATICALLY during klipper restart! Now I understand the post completely ;)
You don't need to run anything on the console, and there is no output to be expected. And no actual tuning through running some movements through the steppers. "Just" a usage/application of settings based on stepper data.

iz3man, 2024-01-23:
Btw: The "tuning" works perfectly fine. My screeching z-axis is now super smooth and super silent. Very happy :)

michaelmbms, 2024-01-24:
The readme could be better explained.
The plugin works indeed perfectly fine.

iz3man, 2024-01-24:
> The readme could be better explained. The plugin works indeed perfectly fine.

If you want me to write some addition that even makes it useable for dummies like me, let me know :)

Superflyron, 2024-03-10:
Did I understand it correctly, I don't have to do anything other than install and activate the script?
I don't have to change anything in my printer.cfg except include_autotune_tmc.cfg?

iz3man, 2024-03-10:
Yes. And the important part (that is not handled or explained in the README) is to add: tuning_goal = auto OR silent OR performance after the entry of each stepper motor.

Superflyron, 2024-03-10:
I had already done that, all my engines are up tuning_goal = Performance . 
I don't really notice a difference now, stealthchop is set to 0, interpolate is set to true.