andrewmcgr/klipper_tmc_autotune #122: Error Could not find motor definition '[motor_constants fysetc-17hs19-2004s-c]' despite being in config

WaschBaer0395, 2024-01-04:
Error Could not find motor definition '[motor_constants fysetc-17hs19-2004s-c]' 
since the latest update i get this error, the motor is 100% in the config 
[image]
[image]
[image]

SvenANT, 2024-01-04:
Hi, I had the same issue, don't use capital letters in your [motor_constants xxx] name or in your "motor: xxx". I was pulling my hair out too :)

ScorpSoul, 2024-01-05:
Same, changes all to lowercase on the database and it worked

WaschBaer0395, 2024-01-05:
yea someone messed up and didnt check if the database still works after they removed whatever converted all names to be lowercase ( why remove that anyways .. dont see a reason to remove that )

closed 2024-01-19 by WaschBaer0395