andrewmcgr/klipper_tmc_autotune #129: I feel a little stupid: But how do you use it?

iz3man, 2024-01-21:
I installed the macros, configured the motors in tmc_autotune.cfg, can call "AUTOTUNE_TMC STEPPER=stepper_x", but nothings happens. Or: What is supposed to happen? I was expecting some kind of stepper movement or something?! What is to be expected.
And how to use it? Do I need to put some commands in my START macro? Or does the tool alter some printer.cfg file?
Sorry for asking these stupid sounding questions, but I don't see them answered. (And not asked, which makes me feel even more stupid ;) ).
Thanks

iz3man, 2024-01-22:
#116

Superflyron, 2024-03-10:
Hello the linked tread is closed, so unfortunately I have to ask here again... So you have to install this tool a tmc_autotune. Create Cfg with the right motors, which are then in the printer. Cfg inklude_tmc_autotune. Cfg . After that, I don't have to activate anything anymore or right?

iz3man, 2024-03-10:
You have to add the following (adaopted to your stepper motors and preference) to your printer.cfg or make an autotune.cfg and include that in printer.cfg. Then restart. And that's it. If you're unsure if it did get applied you can always change some value from "silent" to "performance" and you should note the difference:

```
[autotune_tmc stepper_x]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

[autotune_tmc stepper_y]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

[autotune_tmc stepper_z]
motor: moons-ms17hd6p420I-05
tuning_goal: silent

....
....
```

locki-cz, 2024-03-10:
check klipper logs and you can see detailed configuration for your motors from autotune ...

closed 2025-06-24 by mjonuschat