andrewmcgr/klipper_tmc_autotune #130: Dual-Z on a single driver

mverrilli, 2024-01-26:
Hi, how does this handle dual-Z on a single driver (such as the BTT SKR mini E3 v3)?  I was previously providing extra current as two motors were running off one driver. Does this override the current setting I have in the tmc configuration?

justmaiklas, 2024-06-04:
bump.

Also interested

gammeflamme, 2024-10-16:
I'm no expert but the z steppers on my artillery sidewinder x2 was wired in series so I used 1x max current and 2x the inductance and resistance. I think the BTT SKR mini E3 v3 wires them in parallel so using 0.5x inductance and resistance and 2x max current would equate to the total inductance and resistance with each stepper getting 1x max current

mverrilli, 2024-10-16:
@gammeflamme This makes sense to me. I wonder if the board does anything to automatically compensate based on how many Z steppers are plugged in. I wonder if @bigtreetech would weigh in. 

I've been running it with 1x inductance and resistance, 1.5x max current without any issues, heat, etc. It's just the Z so not a whole lot going on there.

strayr, 2024-12-10:
Steppers are in parallel if you plug them both into the SKR mini. Resistance and inductance both distribute in parallel, but it's not quite that simple as if you have different loads you'll have different back EMF and the load due to the inductance will change. You probably can't run twice the current safely.

gammeflamme, 2024-12-10:
> Steppers are in parallel if you plug them both into the SKR mini. Resistance and inductance both distribute in parallel, but it's not quite that simple as if you have different loads you'll have [...]

Cause the "load" might not be the same on both steppers so one might get more than half?

mjonuschat, 2025-07-31:
I’m closing this old issue as it looks resolved