andrewmcgr/klipper_tmc_autotune #141: Z-motors now make noise

shoikan, 2024-03-08:
Been using autotune for months now to full satisfaction. But after updating to commit 1de500c, my Z motors are making a lot more noise. For one of my Voron 2.4 Z motors, from my printer.cfg:

`[stepper_z]
# Motor M3
# DIR  PB7
# STEP PB8
# EN   PE0
# CS   PB9
# DIAG PF2
step_pin: PB8
dir_pin: PB7
enable_pin: !PE0
rotation_distance: 40
gear_ratio: 80:16
microsteps: 64
endstop_pin: probe:z_virtual_endstop
position_max: 290
position_min: -5
homing_speed: 7
second_homing_speed: 7
homing_retract_dist: 3

[tmc2209 stepper_z]
uart_pin: PB9
interpolate: True
run_current: 0.7

[autotune_tmc stepper_z]
motor: moons-ms17hd6p420I-04
tuning_goal: silent`

The rest are configured similarly. This has been like that for months now, but only after updating to the commit mentioned above it starts making some serious noise when moving up and down slowly, for example when using the tap probe. At higher speeds (when moving the head out of the way for swapping nozzles or something similar) the noise is a lot less.

The A/B motors (same motors, similar config, except set to performance mode) do not appear to suffer from this problem.

TomPcz, 2024-03-10:
I have the same problem with ldo-42sth48-2004ac (tuning_goal: silent) on Z axis of my Voron 2.4. They were very silent before this update. The A/B (ldo-42sth48-2004mah, performance) seem to be fine.

shoikan, 2024-03-10:
Commit 4f1159c improves things somewhat for the Z motors, but overall it's not yet back to the 'old' noise levels. I have add significant amounts of extra_hysteresis to combat vibration/noise on A/B and extruder motors.

shoikan, 2024-04-15:
I feel like commit 6463d8a has fixed my issues. Printer is back to being its quiet self again!

closed 2025-06-25 by mjonuschat