andrewmcgr/klipper_tmc_autotune #144: Constant layer shifts after commit 1d71610a4c61e47d07bca9dfe8dcc9041900c543

celtare21, 2024-03-11:
Using this in config:

```
[autotune_tmc stepper_x]
motor: ldo-42sth48-2004mah
extra_hysteresis: 4

[autotune_tmc stepper_y]
motor: ldo-42sth48-2004mah
extra_hysteresis: 4

[autotune_tmc extruder]
motor: ldo-36sth17-1004ahg
extra_hysteresis: 1
```

I can't finish a single print because it just layer shifts randomly in the first 4 layers. Reverting to commit 1d71610a4c61e47d07bca9dfe8dcc9041900c543 solved all the issues so somewhere till 8db3476afd6a64db1d02218fbb92a8bb77da37cf something broke

WizardPC, 2024-03-13:
I think i have the same issue, it seems that after each Z hop, Z level don't comeback to the same height and drift more and more.. resulting in a failed print.

celtare21, 2024-03-15:
For me it's not from Z-hop (I don't think). I have no Z stepper with autotune_tmc, i just get a random XY layer shift between layers 2 and 6

mjonuschat, 2025-06-25:
Catching up on old issues, is this still happening with the latest autotune version?

mjonuschat, 2025-09-08:
@celtare21 I've exposed the parameters that got changed in the commit as config options for the motors so you can adjust the settings as needed.

commit 775b87cf (2025-09-08): feat: Expose tunables for CoolStep Load Adaptive Current Scaling

CoolStep is an automatic smart energy optimization for stepper motors based on the motor mechanical load, making
them “green.” Depending on the actual chopper mode, CoolStep automatically uses StallGuard4 load measurement
result in StealthChop2, or StallGuard2 in SpreadCycle. Coolstep requires that either StallGuard2 or StallGuard4
(depending on the chopper mode being used) be tuned prior to use. A single tuning does not cover all operating points.

This PR allows adjusting the defaults in TMC autotune (which work well in most u [... cut]
PR #287 (2025-09-08): feat: Expose tunables for CoolStep Load Adaptive Current Scaling
CoolStep is an automatic smart energy optimization for stepper motors based on the motor mechanical load, making
them “green.” Depending on the actual chopper mode, CoolStep automatically uses StallGuard4 load measurement
result in StealthChop2, or StallGuard2 in SpreadCycle. Coolstep requires that either StallGuard2 or StallGuard4
(depending on the chopper mode being used) be tuned prior to use. A single tuning does not cover all operating points.

This PR allows adjusting the defaults in TMC autotune (which work well in most us [... cut]
commit d119f777 (2025-09-08): feat: Expose tunables for CoolStep Load Adaptive Current Scaling

CoolStep is an automatic smart energy optimization for stepper motors based on the motor mechanical load, making
them “green.” Depending on the actual chopper mode, CoolStep automatically uses StallGuard4 load measurement
result in StealthChop2, or StallGuard2 in SpreadCycle. Coolstep requires that either StallGuard2 or StallGuard4
(depending on the chopper mode being used) be tuned prior to use. A single tuning does not cover all operating points.

This PR allows adjusting the defaults in TMC autotune (which work well in most u [... cut]
closed 2025-09-08 by mjonuschat