andrewmcgr/klipper_tmc_autotune #145: Updated 3/11/2024, sensorless homing no longer works for X and Y

yebo29, 2024-03-11:
I'm unsure what it causing it, but I can only now successfully home my machine if I disable tmc_autotune for the X and Y motors. This used to work prior to updating today. Here's the relevant parts of my config:
```
# ALLOW US TO MOVE WITHOUT HOMING
[force_move]
enable_force_move: true

#[autotune_tmc stepper_x]
#motor: omc-17hs19-2004s1
#sg4_thrs: 97
#tuning_goal: auto

#[autotune_tmc stepper_y]
#motor: omc-17hs19-2004s1
#sg4_thrs: 92
#tuning_goal: auto

[autotune_tmc stepper_z]
motor: creality-42-40
tuning_goal: auto

[autotune_tmc extruder]
motor: creality-42-40
tuning_goal: performance

[stepper_x]
step_pin: PB13
dir_pin: PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_min: -1
position_endstop: -1
position_max: 230
homing_speed: 40
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 0.580
stealthchop_threshold: 0
diag_pin: ^PC0
driver_SGTHRS: 97 # 255: most sensitive, 0 is least sensitive
interpolate: true

[stepper_y]
step_pin: PB10
dir_pin: PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_min: 0
position_endstop: 255
position_max: 255
homing_speed: 40
homing_retract_dist: 0
homing_positive_dir: true

[tmc2209 stepper_y]
uart_pin: PC11
tx_pin: PC10
uart_address: 2
run_current: 0.580
stealthchop_threshold: 0
diag_pin: ^PC1
driver_SGTHRS: 92 # 255: most sensitive, 0 is least sensitive
interpolate: true

[stepper_z]
step_pin: PB0
dir_pin: !PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
position_min: -1
position_max: 180

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.8
stealthchop_threshold: 999999

[extruder]
step_pin: PB3
dir_pin: PB4
enable_pin: !PD1
microsteps: 16
gear_ratio: 50:17
rotation_distance: 23.0
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PA0
min_temp: 0
max_temp: 300
max_extrude_only_distance: 610
pressure_advance: 0.33 # Elegoo Black PLA+
#pressure_advance: 0.65 # Overture Red PLA
max_extrude_cross_section: 5 # IS THIS STILL NEEDED?
#min_extrude_temp: 0 # ONLY FOR TESTING

[tmc2209 extruder]
uart_pin: PC11
tx_pin: PC10
uart_address: 3
run_current: 0.75
hold_current:0.375
stealthchop_threshold: 999999
```
Nothing has changed with my configuration prior to the update. No motor changes, no wiring changes, no changes to printer config besides me having to comment out the X and Y tmc_autotune blocks. 

SKR Mini E3 V3 board on The 100 DIY printer using the Extended mod to use Ender 3 bed.

## Symptoms

When homing X, sometimes it goes the wrong way. It'll go about 1-2 mm to the left, and then immediately go to the right and stay there instead of going to safe z_home coordinates. 
Other times it'll go the right way but stop short of where it's supposed to, so homing is incorrect.

When homing Y, it hasn't gone the wrong way for me, but it seems sensitivity is messed up somewhere bc it stops before hitting anything as well.

Disabling tmc_autotune for those results in a successful and correct homing sequence. Let me know what else I can provide to assist troubleshooting.
[image]

Did an update add/remove something that I missed, perhaps?

MLeal91, 2024-03-13:
same problem here with voron 0. sensorless dont work after update

cplagz, 2024-03-14:
Another here, same issue.

b00gie-dev, 2024-03-15:
Same for me. I'm not able to sensorless homing X axis on my V2.4. I had to change my values from 170 down to 50 and still it is not reliable.

cplagz, 2024-03-17:
> Another here, same issue. I've had to significantly drop my values to get it working again (120x 160y down to 45x 60y)

It's working again now.  Here are the steps I took that fixed it.
Updated Klipper to latest
Removed klipper_tmc_autotune folder from Pi
Commented out tmc_autotune section from printer.cfg
Restarted moonraker
Restarted klipper
Re-ran installer script on Pi
restarted moonraker
Uncommented tmc_autotune section
Save + Restart

yebo29, 2024-03-18:
> It's working again now. Here are the steps I took that fixed it. Updated Klipper to latest Removed klipper_tmc_autotune folder from Pi Commented out tmc_autotune section from printer.cfg Restarted [...]

Interesting. I'll try this out later today and report back.

yebo29, 2024-03-18:
Unfortunately that did not fix it for me.

MLeal91, 2024-03-21:
it worked for me. I deleted all printer.cfg data about tmc autotune and moonraker.conf as well. I deleted the installation folder. I did the installation again, and put the settings I had back in and now it works. guys try to do this

b00gie-dev, 2024-04-08:
I personally just stopped using it. Unfortunately. But the motion system is something I have to rely on! 😕

johonan, 2024-04-14:
I also have problems with sensorless with autotune. This is on a voron v0, with the most standard setup with omc motors omc-14hs20-1504s and tmc 2209 drivers. Sensorless homing works fine with autotune disabled, but stops working if I enable it. I tried various values of sg4_thrs but still had no luck.

Config with autotune:
```
[autotune_tmc stepper_x]
motor: omc-14hs20-1504s
tbl: 2
toff: 3
sg4_thrs: 55
```
Config without autotune (relevant parts):
```
[stepper_x]
microsteps: 128
homing_speed: 20

[tmc2209 stepper_x]
run_current: .70
stealthchop_threshold: 0

driver_SGTHRS: 55
driver_TBL: 2
driver_TOFF: 3
driver_HSTRT: 7
driver_HEND: 7
```
klippy.log:
```
autotune_tmc set stepper_x pwm_freq=2
autotune_tmc stepper_x ncycles=219 pfdcycles=-57
autotune_tmc set stepper_x tbl=2
autotune_tmc set stepper_x toff=3
autotune_tmc seting hysteresis based on 24.0 V
dcoilblank = 0.018947, dcoilsd = 0.009333
hysteresis = 13, htotal = 13, hstrt = 8, hend = 5
autotune_tmc set stepper_x hstrt=7
autotune_tmc set stepper_x hend=8
autotune_tmc set stepper_x sgthrs=55
autotune_tmc using max PWM speed 219.388968
autotune_tmc set stepper_x pwm_autoscale=True
autotune_tmc set stepper_x pwm_autograd=True
autotune_tmc set stepper_x pwm_grad=12
autotune_tmc set stepper_x pwm_ofs=31
autotune_tmc set stepper_x pwm_reg=15
autotune_tmc set stepper_x pwm_lim=4
autotune_tmc set stepper_x tpwmthrs=1048575
autotune_tmc set stepper_x en_spreadcycle=True
autotune_tmc set stepper_x tcoolthrs=391(24.0)
autotune_tmc set stepper_x semin=2
autotune_tmc set stepper_x semax=2
autotune_tmc set stepper_x seup=3
autotune_tmc set stepper_x sedn=2
autotune_tmc set stepper_x seimin=0
autotune_tmc set stepper_x iholddelay=12
autotune_tmc set stepper_x multistep_filt=True
```
tmc dump without autotune:
```
$ DUMP_TMC stepper=stepper_x
// ========== Write-only registers ==========
// SLAVECONF: 00000200 senddelay=2
// IHOLD_IRUN: 00081616 ihold=22 irun=22 iholddelay=8
// TPWMTHRS: 000fffff tpwmthrs=1048575
// TPOWERDOWN: 00000014 tpowerdown=20
// SGTHRS: 00000037 sgthrs=55
// TCOOLTHRS: 00000000
// ========== Queried registers ==========
// GCONF: 000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
// GSTAT: 00000000
// IFCNT: 0000003c ifcnt=60
// OTP_READ: 0000000e otp_fclktrim=14
// IOIN: 210002c9 enn=1 ms2=1 pdn_uart=1 step=1 dir=1 version=0x21
// FACTORY_CONF: 0000000e fclktrim=14
// TSTEP: 000fffff tstep=1048575
// MSCNT: 00000273 mscnt=627
// MSCURACT: 0144015f cur_a=-161 cur_b=-188
// CHOPCONF: 310303f3 toff=3 hstrt=7 hend=7 tbl=2 vsense=1 mres=1(128usteps) intpol=1 dedge=1
// DRV_STATUS: 80160000 cs_actual=22 stst=1
// PWMCONF: c80d0e24 pwm_ofs=36 pwm_grad=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 pwm_reg=8 pwm_lim=12
// PWM_SCALE: 00000023 pwm_scale_sum=35
// PWM_AUTO: 001b0032 pwm_ofs_auto=50 pwm_grad_auto=27
// SG_RESULT: 00000000
```

johonan, 2024-04-14:
I found an easy way to reproduce the problem without autotune: run
```
SET_TMC_FIELD stepper=stepper_x field=tcoolthrs value=391
```
and try to home. Homing will not work

johonan, 2024-04-14:
And sensorless homing works with autotune if I run
```
SET_TMC_FIELD stepper=stepper_x field=tcoolthrs value=490
SET_TMC_FIELD stepper=stepper_y field=tcoolthrs value=490
```
It fails with tcoolthrs below 480 and is potentially unstable between 480 and 490.

I have no idea what tcoolthrs is, and what these values are supposed to measure!

johonan, 2024-04-14:
I found settings when sensorless homing works with autotune in my case:
```
[autotune_tmc stepper_x]
motor: omc-14hs20-1504s
tbl: 2
toff: 3
sg4_thrs: 140
```
```
[stepper_x]
homing_speed: 80
```
This is confusing because I tried bumping up homing speed and various values of `sg4_thrs` before,  but hell, it works!

locki-cz, 2024-04-14:
this is with my tmc2240 and latest autotune:

[autotune_tmc stepper_x]
motor: moons-ms17hd6p420I-05
voltage: 36
sgt: 1

[autotune_tmc stepper_y]
motor: moons-ms17hd6p420I-05
voltage: 36
sgt: 1

and this is with tmc5160 plus also on latest and works:

[autotune_tmc stepper_x]
motor: moons-ms17hd6p420I-05
voltage: 48
sgt: 1

[autotune_tmc stepper_y]
motor: moons-ms17hd6p420I-05
voltage: 48
sgt: 1

[tmc5160 stepper_x]
cs_pin: PC4
spi_software_miso_pin: PA6
spi_software_mosi_pin: PA7
spi_software_sclk_pin: PA5
interpolate: true
stealthchop_threshold: 0
sense_resistor: 0.022
run_current: 0.8 #0.976   #0.7 0.976 
#sense_resistor: 0.110
diag1_pin: !PG6 # use the same pin that was previously the endstop_pin!
driver_SGT: 1   #sensorless homing

plus homing maclo for lower current to drivers:

```
[gcode_macro _HOME_X]
gcode:
    # Always use consistent run_current on A/B steppers during sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc5160 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc5160 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.4 %}
    #SET_TMC_FIELD FIELD=SGTHRS STEPPER=stepper_x VALUE=90
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}

    # Home
    G28 X
    # Move away
    G91
    G1 X-10 F1200
    G90
    
    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}

[gcode_macro _HOME_Y]
gcode:
    # Set current for sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc5160 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc5160 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.4 %}
    #SET_TMC_FIELD FIELD=SGTHRS STEPPER=stepper_y VALUE=90
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}

    # Home
    G28 Y
    # Move away
    G91
    G1 Y-10 F1200
    G90

    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
```

yebo29, 2024-04-20:
Just updated to latest (v0.2.0-165-g6463d8ae) and tried again and it worked on the first attempt. Will require further testing to make sure it's not just a one-off.

yebo29, 2024-04-27:
So far it has continued to remain solid after the previous update with no further interventions. Are we good to close this one or are others still seeing an issue after updating?

yebo29, 2024-05-22:
It broke again after updating today. Looked at another opened [issue](https://github.com/andrewmcgr/klipper_tmc_autotune/issues/12#issuecomment-2125532084), updating homing speed for z and y seems to have fixed it for now. I did reinstall beforehand.

Lekvaran, 2024-05-23:
For me not work too.

numanair, 2024-05-25:
2209's and I had to increase my homing current to get enough sensitivity (especially on Y). I have not tried adjusting the speed yet. I'm quite curious about the real cause here!

mjonuschat, 2025-06-25:
Catching up on old issues, is this still happening on the latest autotune version?

mjonuschat, 2025-07-31:
I’m closing this old issue. If it’s still happening, please open a new one

closed 2025-07-31 by mjonuschat