andrewmcgr/klipper_tmc_autotune #148: Data for creality 42-60

pauluchin, 2024-03-23:
Hello, I have an ender 7 but I can't find information about the motors in the cfg. If anyone had information about the 42-60 it occupies, I would appreciate it.

bra1nDeaD, 2024-03-30:
Creality uses sizes to identify their motors (i.e: 42-60). But this is not the actual type identifier of the motor. Have a look on the bottom of the motor and see if you can find a proper identifier on it. Once you have that you need to find the datasheet for it. You can ask the manufacturer (Creality doesn't manufacture motors themself) directly for a datasheet. This should contain all the information you need. Once you've found them, post them here so they can be added to the database.

Key-Alex, 2024-03-31:
I'm now trying to add this motor.
[motor_constants creality 42-60] 
#Creality Ender 7 - RB Step Motor 17HDC8033-32B
# Coil resistance, Ohms
resistance: 1.2
# Coil inductance, Henries
inductance: 0.0034
# Holding torque, Nm
holding_torque: 0.76
# Nominal rated current, Amps
max_current: 2.00
# Steps per revolution (1.8deg motors)
steps_per_revolution: 200

mjonuschat, 2025-06-24:
Closing as the motor is in the motor database as `creality-42-60`

closed 2025-06-24 by mjonuschat