andrewmcgr/klipper_tmc_autotune #149: Commit 4f1159c Issues

Ibot-11, 2024-03-24:
Commit 4f1159c causes a lot of issues.
First the sensorless homing sensitivity changed and my voron does not home properly anymore with the same settings.
I can also no longer reach 500mm/s without insane step losses.
1de500c works without any issues

Tested on my Voron 2.4R2 with ldo-42sth48-2504mac

dhruvinsh, 2024-03-29:
likewise, I have lost some precision, still trying to bisect though what causing this.

SonnyRazzvan, 2024-04-02:
Having the same issue with the latest update.

mhaas, 2024-05-20:
Adding a me too - with latest HEAD, I was getting skipped steps even at 5k accel and 200mm/s speed. Switching to rev https://github.com/andrewmcgr/klipper_tmc_autotune/commit/1de500cff54db2400c9a1fd52b892d901f024a33 fixed that!

mjonuschat, 2025-06-24:
Is this still an issue on the lasted version?

closed 2025-07-31 by mjonuschat