andrewmcgr/klipper_tmc_autotune #154: Sensorless homing issue with 2240

rfnovo, 2024-04-09:
Got sensorless homing working fine with FYSETC TMC2240 and the following config:
[tmc2240 stepper_x] 
# Soft SPI
spi_software_mosi_pin: PB15
spi_software_miso_pin: PB14
spi_software_sclk_pin: PB13
cs_pin: PC11
diag0_pin: ^!PB3
interpolate: False
run_current: 0.7
driver_SGT: 3
stealthchop_threshold: 0

When adding Autotune it seems to ignore the sgt parameter and always slam the toolhead on the gantry until motor skip steps:
[autotune_tmc stepper_x]
motor: fysetc-35hsh7402-24b-60a
sgt: -60

For reference previsouly, with 2209 sensorless worked fine with autotune

bananensoftware, 2024-04-24:
Same Issue with LDO Leviathan TMC5160 A/B Drivers.

locki-cz, 2024-04-24:
Then probably wrong diag 0 or 1 pin

bananensoftware, 2024-04-24:
I got it working by disabling TMC Autotune. With autotune acitve it slams and skips steps until I hit the emergency stop.

cromine11, 2024-05-01:
What's your homing speed for the 2240?

rfnovo, 2024-05-01:
Usually 40, but tried with as low as 20. Without Autotune it works fine, add autotune and it crashes. Also even using extreme values like sgt -60 seems to have no effect. Toolhead should not even move with such low value. Looks like it is just ignoring the SGT value at all

EricZimmerman, 2024-06-26:
You need the driver value IN THE AUTORUNE SECTION just like the docs say. 

Why would you have 3 in your config block (that autotune ignores) and -60 in the autotune section??

EricZimmerman, 2024-06-26:
What is homing speed and homing current?

I have working 2240 examples on my GitHub sensorless guide

EricZimmerman, 2024-06-27:
https://github.com/EricZimmerman/VoronTools/blob/main/Sensorless.md

Start there

mjonuschat, 2025-06-24:
Has this been resolved using the linked guide?

bananensoftware, 2025-06-24:
Tried it some time ago and it worked, same config as before. Seems to be solved by an updated.