andrewmcgr/klipper_tmc_autotune #168: Little clarification on former stepper settings

krzyzankowskig, 2024-05-05:
Hey, 
I'm trying to find out what parameters I should remove from my [tmc2209 stepper_*] sections. I already found out (the hard way...) that I need to remove driver_SGTHRS fro sensor-less homing. But what other settings should be removed? Or are safe to stay? Like is stealthchop_threshold taking any effect or being overwritten, microsteps etc. 

My Z is very loud after autotuning and I'm afraid I left something in config that's messing up with autotuning.

ps. This is a really great project :) X and Y are working smooth as butter (and quiet). Kudos for author.

mjonuschat, 2025-06-24:
Does the [Adjusting existing configuration](https://github.com/andrewmcgr/klipper_tmc_autotune?tab=readme-ov-file#adjusting-existing-configuration) section in the readme help? If not, how can we make it clearer?

mjonuschat, 2025-07-31:
I’m closing this old issue.

closed 2025-07-31 by mjonuschat