andrewmcgr/klipper_tmc_autotune #172: Section 'autotune_tmc stepper_x' is not a valid config section

TheVoronModder, 2024-05-18:
I keep getting errors:

Section 'autotune_tmc stepper_x' is not a valid config section

Here is my config section:

```
[autotune_tmc stepper_x]
motor: ldo-42sth48-2004mah

[autotune_tmc stepper_x1]
motor: ldo-42sth48-2004mah

[autotune_tmc stepper_y]
motor: ldo-42sth48-2004mah

[autotune_tmc stepper_y1]
motor: ldo-42sth48-2004mah

[autotune_tmc extruder]
motor: ldo-36sth20-1004ahg
```

dezenten, 2024-06-08:
Check the name of the steppers and adjust so it matches your config. If you have a  Voron you only have one of X & Y.

LordKillZone, 2024-10-15:
I have the same problem, after update it showed up same stpper_x error, before update it was working on the same config.

ImShogun, 2025-01-13:
Hi there, sound very interesting to me but indeed I cant run the printer with any "autotune_tmc ***" block.

I see in Mainsail "klipper_tmc_autotune v0.2.0-240-g9f0777e1" "up to date" and I simplified my tests to just the extruder, meaning I have those blocks for them:

`[extruder]
step_pin: P1.15
dir_pin: P1.14
enable_pin: !P1.16
microsteps: 64 #16
nozzle_diameter: 0.400
filament_diameter: 1.750
full_steps_per_rotation: 200
rotation_distance:24.62 #microswiss default:24.62  #dualgear extruder black before microswiss23.64 #11.082 #13.83 #23.520
max_extrude_only_distance: 600
heater_pin: P2.7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: -273.15
max_temp: 250
min_extrude_temp: 0
#<start> + <measured_height> * <factor>. (For example, 0 + 12.90 * .020 would be .258.)
#result for Benoit CR10S: 26mm --> 0 + 26 * 0.020 = 0.52
#pressure_advance: 0.52

[tmc2209 extruder]
uart_pin: P1.1
run_current: 0.950
hold_current: 0.400
# https://github.com/KevinOConnor/klipper/issues/1363
#stealthchop_threshold: 5000
interpolate: true

[autotune_tmc extruder]
motor: oukeda-17hs4401`

To bad as indeed it looks promising. My CR10S, since updated to klipper (including a 32bits mb I had around whilst the original fried) have those "zebra lines" in some spots, and I was running short of ideas on how to fix them :-)

translucentfocus, 2025-06-21:
Seems this is a repeat of this issue: https://github.com/andrewmcgr/klipper_tmc_autotune/issues/99

Fix is to re-install ```wget -O - https://raw.githubusercontent.com/andrewmcgr/klipper_tmc_autotune/main/install.sh | bash```

Have tested this fix myself on latest Raspberry Pi OS.

mjonuschat, 2025-06-24:
Closing this out as the fix has been mentioned. This error can occur when Klipper was reported as "corrupted" by Mainsail/Fluidd and a "Recovery" was initiated from the WebUI / Klipper was force updated.

Fix is to re-run the installer.