andrewmcgr/klipper_tmc_autotune #173: no attribute 'get_step_dist' after recent Klipper commit

sprior, 2024-05-22:
I believe the commit at https://github.com/Klipper3d/klipper/commit/faee2c0e52506f157d38b747613df213d659226a
broke compatibility with TMC Autotune.  After updating to latest Klipper I started getting an error no attribute 'get_step_dist'.

Commenting out my autotune_tmc lines in printer.cfg seems to have resolved this temporarily.

Do we need a compatibility update?

PendulumPit, 2024-05-22:
`Internal error during ready callback: 'float' object has no attribute 'get_step_dist'
This is what I get unless I comment out the TMC Autoune

jrgiacone, 2024-05-22:
git reset --hard 694d38c79105f2d33242487faa0532d1291ee7b6 to fix it, i believe this refactor broke compatibility with tmc autotune: https://github.com/Klipper3d/klipper/commit/faee2c0e52506f157d38b747613df213d659226a

matthewj301, 2024-05-22:
same here, commented out for now until a compatibility patch is released

nhchiu, 2024-05-22:
Ran into the same issue.
Seems to be fixed in 436f6f19a0af53a5b869c250e932e9202f3595d8.

notheotherben, 2024-05-22:
Just a warning here that updating to the latest release required re-tuning stallguard for sensorless homing (on TMC2209s in my case). Be careful folks....

sbtoonz, 2024-05-27:
https://github.com/andrewmcgr/klipper_tmc_autotune/blob/main/autotune_tmc.py#L261

calls get_step_dist() from stepper 

but

https://github.com/Klipper3d/klipper/blob/master/klippy/stepper.py#L108-L109

that function is exported here

is the variable being called improperly now since refactor @andrewmcgr

sbtoonz, 2024-05-27:
https://github.com/andrewmcgr/klipper_tmc_autotune/commit/436f6f19a0af53a5b869c250e932e9202f3595d8

This commit closes this issue