andrewmcgr/klipper_tmc_autotune #174: Internal error during connect: cannot import name signature

h3rm, 2024-05-24:
**Describe the bug**
Internal error during connect: cannot import name signature error message. If i comment out all autotune rows in the printer.cfg the printer can suddenly connect again

**To Reproduce**
Steps to reproduce the behavior:
after todays full update i get the error:
https://imgur.com/cBKVtbn

**Expected behavior**
normal start

**Screenshots**
https://imgur.com/cBKVtbn

**Logfile**
[moonraker.log](https://github.com/Guilouz/KlipperScreen-Flsun-Speeder-Pad/files/15426951/moonraker.log)
[klippy.log](https://github.com/Guilouz/KlipperScreen-Flsun-Speeder-Pad/files/15426953/klippy.log)

h3rm, 2024-05-24:
> 5hagbard23 found the solution, and Posted it in discord. Python is too old and should be Updated : `sudo systemctl stop klipper; rm -rf ~/klippy-env; python3 -m venv ~/klippy-env; ~/klippy-env/bin/pip install -r ~/klipper/scripts/klippy-requirements.txt; sudo systemctl start klipper` works on my side

maybe use with care. i bricked my klipper config totally with this

vndk36, 2024-06-15:
I had the same issue and the klippy logs where not super clear. I tried the commands on the command line. It worked well on my side without bricking the klipper config. Your case may vary. Thx @Dragi2k