andrewmcgr/klipper_tmc_autotune #187: Configuration difficulty..

hreality, 2024-07-17:
Please, overlook my ignorance.. could you update your instructions and include a functioning example for X/Y (drivers, steppers etc) printer.cfg.  Im having issues and my searches cant seem to find a working model as an example.

eefweenink, 2024-07-20:
Yesterday I installed autotune for the first time. Was also trying my way, by trial and error. 
Very helpfull was to just open https://github.com/andrewmcgr/klipper_tmc_autotune/blob/main/motor_database.cfg 
And search there for the motors on my machine. Copy-pasted them to the printercfg. Done

mjonuschat, 2025-07-31:
Closing this old issues. The README has a (minimal) [config example](https://github.com/andrewmcgr/klipper_tmc_autotune?tab=readme-ov-file#autotune-configuration) included and all optional parameters documented below it.

closed 2025-07-31 by mjonuschat