andrewmcgr/klipper_tmc_autotune #190: Why is there no Guide how to use this?

Durahl, 2024-07-29:
- I've installed the Git
- I've added the Update Code to Moonraker
- I've added the missing hold_current to the tmc_5160 stepper_x/y sections together with the motor_constants and autotune_tmc Code to printer.cfg

**Now what?**
Just running `AUTOTUNE_TMC STEPPER=stepper_x` in the Console does nothing - No Movements, no Reports, nothing like what you'd experience when doing a Resonance Calibration... Not even an Error Message.

What are we supposed to be looking for? Would someone please give an explanation how to operate this to someone with ZERO knowledge how to operate this?

dagolard, 2024-08-09:
Did you add your steppers to the printer.cfg?

i.e. like mine...

[autotune_tmc stepper_x]
motor: ldo-42sth48-2004mah
voltage: 48

[autotune_tmc stepper_x1]
motor: ldo-42sth48-2004mah
voltage: 48

[autotune_tmc stepper_y]
motor: ldo-42sth48-2004mah
voltage: 48

[autotune_tmc stepper_y1]
motor: ldo-42sth48-2004mah
voltage: 48

[autotune_tmc stepper_z]
motor: ldo-42sth60-2004ac
voltage: 24

[autotune_tmc stepper_z1]
motor: ldo-42sth60-2004ac
voltage: 24

[autotune_tmc stepper_z2]
motor: ldo-42sth60-2004ac
voltage: 24

[autotune_tmc extruder]
motor: ldo-36sth20-1004ahg
voltage: 24

Durahl, 2024-08-10:
Well yea, to _some_ degree ( since there is no Example Guide ):
[image]

What are we supposed to do after entering all the Edits into the `printer.cfg`?

Xplode85, 2024-08-17:
After you do the edits there isn't really anything else to do except notice the different sounds from your steppers.
There is no real active autotuning, it's just setting stepper driver parameters based on the stepper motors you have added in `printer.cfg`

This has worked succesfully for me and all my printers got quieter, stepper drivers did get a little hotter thou so i got a temperature warning.

Durahl, 2024-08-17:
Thanks for the Clarification.

Unfortunately, it didn't seem to do a lot ( if anything? 🤔 ) for me even when adding the remainder Configuration possibilities to it... Sound didn't change when switching from Auto ( Performance for X/Y ) to Silent. Ah well... 😑

Fredlepp, 2024-10-27:
moi aussi , je lance n'importe quelle de ces commandes:
AUTOTUNE_TMC STEPPER=extruder
AUTOTUNE_TMC STEPPER=stepper_x
AUTOTUNE_TMC STEPPER=stepper_y
AUTOTUNE_TMC STEPPER=stepper_z
AUTOTUNE_TMC STEPPER=stepper_z1
AUTOTUNE_TMC STEPPER=stepper_z2
AUTOTUNE_TMC STEPPER=stepper_z3

il n'y a pas de message d'erreur mais rien ne se passe

mjonuschat, 2025-07-31:
I’m closing this old issue.The documentation has a few edits that show how to check that TMC autotune did run on startup. Not every printer/motor combination will have audible changes.

closed 2025-07-31 by mjonuschat