andrewmcgr/klipper_tmc_autotune #191: FR: [manual_stepper] Including manual Steppers support

Firas95k, 2024-08-17:
Hello, thank you for the the awesome work, is it possible to include support to all the defined steppers no matter what the are called in the printer.cfg, including [manual_stepper]'s

orion23rigel, 2024-10-28:
I would also love to have support for manual_steppers

gentlewatcher, 2024-11-16:
May I also ask if we can support extruder_stepper (I am doing a multi extruder setup with syncing to the main extruder object but cant set the autotune for those other stepper drivers).

mjonuschat, 2025-07-31:
I believe this works in the latest version - I don't have a manual_stepper setup, but with this config for an extruder_stepper autotuning is applied based on the log:

**Config**:
```
[extruder_stepper nightowl_xl]
extruder: extruder
step_pin: nxl:MOT0_STEP
dir_pin: nxl:MOT0_DIR
enable_pin: !nxl:MOT0_EN
full_steps_per_rotation: 200
microsteps: 16
rotation_distance: 22.7409
gear_ratio: 4:1

[tmc2209 extruder_stepper nightowl_xl]
uart_pin: nxl:MOT0_CS
run_current: 0.8
stealthchop_threshold: 0
interpolate: True
sense_resistor: 0.110

[autotune_tmc extruder_stepper nightowl_xl]
motor: omc-17hm19-2004s
tuning_goal: silent
```

**Log**:
```
autotune_tmc set extruder_stepper nightowl_xl pwm_freq=2
autotune_tmc extruder_stepper nightowl_xl ncycles=219 pfdcycles=73
autotune_tmc set extruder_stepper nightowl_xl tbl=1
autotune_tmc set extruder_stepper nightowl_xl toff=1
autotune_tmc set extruder_stepper nightowl_xl hstrt=1
autotune_tmc set extruder_stepper nightowl_xl hend=3
autotune_tmc set extruder_stepper nightowl_xl sgthrs=40
autotune_tmc set extruder_stepper nightowl_xl pwm_autoscale=True
autotune_tmc set extruder_stepper nightowl_xl pwm_autograd=True
autotune_tmc set extruder_stepper nightowl_xl pwm_grad=6
autotune_tmc set extruder_stepper nightowl_xl pwm_ofs=18
autotune_tmc set extruder_stepper nightowl_xl pwm_reg=15
autotune_tmc set extruder_stepper nightowl_xl pwm_lim=4
autotune_tmc set extruder_stepper nightowl_xl tpwmthrs=0
autotune_tmc set extruder_stepper nightowl_xl en_spreadcycle=False
autotune_tmc set extruder_stepper nightowl_xl tcoolthrs=78(17.055675)
autotune_tmc set extruder_stepper nightowl_xl semin=2
autotune_tmc set extruder_stepper nightowl_xl semax=4
autotune_tmc set extruder_stepper nightowl_xl seup=3
autotune_tmc set extruder_stepper nightowl_xl sedn=2
autotune_tmc set extruder_stepper nightowl_xl seimin=1
autotune_tmc set extruder_stepper nightowl_xl iholddelay=12
autotune_tmc set extruder_stepper nightowl_xl multistep_filt=True
```

closed 2025-09-07 by mjonuschat