andrewmcgr/klipper_tmc_autotune #192: Is `hold_current` required for auto tune?

spikeygg, 2024-08-18:
I've been looking at tuning the drivers and [according to the klipper docs, hold_current should no longer be used](https://www.klipper3d.org/TMC_Drivers.html#prefer-to-not-specify-a-hold_current). However, the instructions in this GitHub repo's readme say:

> - Currents (run current, **hold current**, homing current if using a Klipper version that supports the latter)

I also found [this post from Sineos on the klipper discourse server](https://klipper.discourse.group/t/hold-current-and-run-current-tmc2209/1171/5) from 2021 which cites that there are negative impacts from using both `run_current` and `hold_current`.

I installed the klipper_tmc_autotune and ran it without adding the `hold_current` to my configuration and everything seems to be working... I just don't want to be missing something important here.

robustini, 2024-08-20:
hold_current is not recommended any longer.
It has been found that the switch between run_current and hold_current has a negative impact on precision.
So this script should not use it at all.

Sonnenwerk, 2025-01-02:
Had the same question this day. Anyone willing to adapt the docs ?

andrewmcgr, 2025-01-02:
The klipper docs are correct *for klipper's default settings* and not
correct for a properly tuned driver. You can deduce this from the chip
datasheets.

If you can demonstrate that I'm wrong with actual data (say, hook up an
encoder to a motor and track what actually happens at adequate precision),
sure, but otherwise it's staying how it is.

Sonnenwerk, 2025-01-05:
Ah ! You are right !

stablestud, 2025-01-23:
What would be the recommended hold_current anyways? I wouldn't know how to size the hold_current correctly