andrewmcgr/klipper_tmc_autotune #206: Logging to console

thijstriemstra, 2024-10-14:
I recently installed KAMP and it logs the following to the Fluidd console when invoked (before every print):

```
16:08:46 // Moving filament tip 0.0mms
16:08:46 // KAMP purge is using firmware retraction.
16:08:46 // KAMP purge starting at 98.0, 91.779 and purging 20.0mm of filament, requested flow rate is 10.0mm3/s.
```

would be nice if this plugin can also log something to the console so I know it's properly intialized.

stablestud, 2025-01-23:
I'm against general output, but rather output only on warnings and errors.
Optionally debug output should be able toggleable, but not by default

mjonuschat, 2025-06-26:
This work for KAMP as that is based on macros and is not a Klipper plugin. The console output only works when at least Moonraker is connected to the Klippy server. As autotune_tmc works during the early klipper startup the moonraker connection is not yet established and thus no messages can/will be logged.

closed 2025-06-26 by mjonuschat