andrewmcgr/klipper_tmc_autotune #208: First layer issues with multiple object on Trident after autotune

MallocArray, 2024-10-19:
Trident 300 cubed LDO Rev A Kit.  
After enabling klipper_tmc_autotune for the motors I have, if I'm printing multiple objects, the very first object's first layer looks normal.  When it moves to the first layer of the second object, it is slightly higher.  When it moves to other objects, it is even higher, eventually to the point that prints will fail.

If I comment out the autotune config section and reprint the same file, the first layer on each item is consistent across the objects.
Unsure if I've implemented something wrong.  I verified the Z-Motor models and used documentation for the other motors in my build.

Latest Klipper, Octopus 1.1 board with BigTreeTech TMC2209-v1.2

```
[autotune_tmc stepper_x]
motor: ldo-42sth48-2004mah
[autotune_tmc stepper_y]
motor: ldo-42sth48-2004mah

[autotune_tmc stepper_z]
motor: ldo-42sth40-1684cl350et
[autotune_tmc stepper_z1]
motor: ldo-42sth40-1684cl350et
[autotune_tmc stepper_z2]
motor: ldo-42sth40-1684cl350et

[autotune_tmc extruder]
motor: ldo-36sth20-1004ahg
```

Objects 1-3 left to right
[image]

MallocArray, 2024-10-21:
Unclear if my issues are related to this potential Klipper bug, exasperated by autotune, or unrelated entirely

https://klipper.discourse.group/t/bug-accumulating-microstep-shift-during-probe-moves-related-to-endstop-oversampling/19429/5

Host(aarch64, 32bit)
Version: v0.12.0-317-gb89d55238

Relevant Klipper config
```yaml
#####################################################################
#   Z Stepper Settings
#####################################################################

##  Z0 Stepper - Front Left
##  Connected to MOTOR_2
##  Endstop connected to DIAG_2
[stepper_z]
step_pin: PF11
dir_pin: !PG3
enable_pin: !PG5
rotation_distance: 4    # Rotation Distance for TR8x8 = 8, TR8x4 = 4, TR8x2 = 2
microsteps: 32
full_steps_per_rotation: 200
endstop_pin: PG10
# position_endstop = -1.300  # For ABS  -1.340 for PLA
position_max: 290
position_min: -2.5
homing_speed: 8.0 # Leadscrews are slower than 2.4, 10 is a recommended max.
second_homing_speed: 3
homing_retract_dist: 3

[tmc2209 stepper_z]
uart_pin: PC6
interpolate: False
run_current: 0.6
sense_resistor: 0.110
stealthchop_threshold: 0

##  Z1 Stepper - Rear Center
##  Connected to MOTOR_3
[stepper_z1]
step_pin: PG4
dir_pin: !PC1
enable_pin: !PA0
rotation_distance: 4  # Rotation Distance for TR8x8 = 8, TR8x4 = 4, TR8x2 = 2
microsteps: 32
full_steps_per_rotation: 200

##  Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z1]
uart_pin: PC7
interpolate: False
run_current: 0.6
sense_resistor: 0.110
stealthchop_threshold: 0

##  Z2 Stepper - Front Right
##  Connected to MOTOR_4
[stepper_z2]
step_pin: PF9
dir_pin: !PF10
enable_pin: !PG2
rotation_distance: 4  # Rotation Distance for TR8x8 = 8, TR8x4 = 4, TR8x2 = 2
microsteps: 32
full_steps_per_rotation: 200

##  Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z2]
uart_pin: PF2
interpolate: False
run_current: 0.6
sense_resistor: 0.110
stealthchop_threshold: 0

[probe]
##  Inductive Probe
##  This probe is not used for Z height, only Quad Gantry Leveling
## For the DIAG_7 port. NEEDS BAT85 DIODE! Change to !PG15 if probe is NO.
pin: ^PG15
x_offset: 0
y_offset: 25.0
z_offset: 0
speed: 10.0
samples: 3
samples_result: median
sample_retract_dist: 3.0
samples_tolerance: 0.006
samples_tolerance_retries: 3

```

WCEngineer, 2025-02-06:
I have also had issues with that same model of motors, I think there is something wrong with the register values that autotune is setting for those motors when `tuning_goal` is set to `silent` (the default for z-axis and extruders). For me, I immediately heard skipped steps on all 3 z-steppers during the travel moves when I tried to home my printer after enabling autotune, and increasing the run_current did not make a difference.

If you want a quick fix, try adding `tuning_goal: performance` to each of your z-steppers, like this:
```
[autotune_tmc stepper_z]
motor: ldo-42sth40-1684cl350et
tuning_goal: performance
[autotune_tmc stepper_z1]
motor: ldo-42sth40-1684cl350et
tuning_goal: performance
[autotune_tmc stepper_z2]
motor: ldo-42sth40-1684cl350et
tuning_goal: performance
```
After switching these motors to `tuning_goal: performance` it worked as expected.

I don't care about the stepper noise, so I'm fine leaving it on `performance`, but the `silent` settings should not be resulting in skipped steps like that

penchochris, 2025-02-15:
I also have problems with z_offset. I had to remove autotune_tmc on Z motors since it was giving me very different values on z_offset using paper test. What I've seen is while using klipper screen to calibrate PROBE_CALIBRATE smaller values caused skipped steps or maybe wrong movement... motor was moving on both directions which is weird...

penchochris, 2025-02-15:
@andrewmcgr

WCEngineer, 2025-02-17:
> NOTE 2: Sounds like theres a problem with LDO motors in Z axis. Checked voron's discord and there's some people having the same problems with motor: ldo-42sth48-2004ac

That model (ldo-42sth48-2004ac) is the stock motor for V2.4 (belted Z) so it typically has a plain shaft with a toothed pulley installed on it, whereas ldo-42sth40-1684cl350et is an integrated lead screw motor, as seen of Trident. The fact that these issues are being reported on both lead screw motors and belted Z may be relevant.

WCEngineer, 2025-06-25:
> Catching up on old issues, is this still happening on the latest autotune version?

I will have to undo my workaround and run some tests, I am a little rusty on this topic.

mjonuschat, 2025-07-31:
> It is indeed still a problem when quickly tested at `run_current: 0.6`, I will now play with the `run_current` to see if higher values can allow normal movement

Any updates with higher run current?

WCEngineer, 2025-07-31:
> Any updates with higher run current?

From what I remember, increasing the run current did not seem to make a difference. I believe I also tried lowering`max_z_velocity` and `max_z_accel`, but I do not recall what values I tried for those. 

It looks like I put it back to `run_current: 0.6` and `tuning_goal: performance` and then forgot about it.

Marucci87, 2025-10-24:
shouldnt interpolate be set to interpolate: true ? Idk if that would cause the issue but I have mine set to true as the documentation says to do

schutztj, 2025-11-23:
This issue is still happening as of now. It is still required to set your tuning goal to performance. Should we switch this label to "bug" instead of question?

Not that familiar with steppers but I could get my z axis homed but then any attempted move in z would cause this grinding sound and the movement wouldn't complete. I tried rolling Kalico back behind the last tmc change which hand no affect. Changing the tuning goal to performance resolved the issue.