andrewmcgr/klipper_tmc_autotune #211: Microsteps setting

krzyzankowskig, 2024-10-30:
Guide says: "Autotune gives better results, both dimensionally and quality, by using interpolation and as many microsteps as feasible". So... Should we set it in config? Or auto tune sets it according to driver and motor specs automatically? 
If we set it, is there any suggested value?

LabunskyA, 2024-12-31:
You should set it according to your driver. TMC 2209 can use max of 64, for example

charredchar, 2025-01-01:
> You should set it according to your driver. TMC 2209 can use max of 64, for example

Are you sure about this? I know the datasheet says a max of 64 when using the dir and step pins but I've tested speeds with every microstep setting between 16 and 256 and found an improvement each step above 64, even at 256. If it was in fact limited to 64 I would be curious to know why 64 was actually one of the worst settings for me.

andrewmcgr, 2025-01-02:
All the drivers autotune works with support up to 256 microsteps. However,
that can easily be more step pulses than your MCU can handle, and if you
run out of MCU processor time you get 'timer too close' errors. This is why
interpolation is useful; the driver still runs at 256 microsteps, but the
MCU doesn't need to.

LabunskyA, 2025-01-02:
> Are you sure about this? I know the datasheet says a max of 64 when using the dir and step pins but I've tested speeds with every microstep setting between 16 and 256 and found an improvement each [...]

You *can* set them to 256. But it may lead to skipped steps depending on your currents and speeds. Maximum *safe* value is 64 with virtual interpolation to 256 steps

krzyzankowskig, 2025-01-02:
And we co e back to my oryginał question: should we set it or autorune is handling it :)

Sonnenwerk, 2025-01-02:
I have the same question, as it is not clear which parameter 
 1. must stay in driver section
 2. should /can stay Driver section
 3. shall not! stay in Driver section

LabunskyA, 2025-01-02:
> And we come back to my oryginał question: should we set it or autorune is handling it :)
> 
> Ps. How you can set number of interpolated steps?

You should step it. Klipper will not let you left microsteps unspecified :)
Set it to the highest safest amount the driver allows. For TMC2209 it's 64 (you can go higher if you personally want to, but it may lead to problems)

The basic config example (taken from flsun v400 with 1.8 motors on TMC2209):
```
[stepper_a]
step_pin: PE3 
dir_pin: PE2
endstop_pin: ^PA15
enable_pin: !PE4
microsteps: 64
rotation_distance: 40

[tmc2209 stepper_a]
uart_pin: PD5
run_current: 1.5
interpolate: true
sense_resistor: 0.110 
```

krzyzankowskig, 2025-01-02:
So if you set 64 micro steps and add interpolate, will it now to interpolate it to 256? How it works is little unclear to me :)

LabunskyA, 2025-01-02:
> So if you set 64 micro steps and add interpolate, will it now to interpolate it to 256? How it works is little unclear to me :)

Basically, in the example above, Klipper calculates movement with assumed 64 microsteps of a driver motor. Then it sends them to the TMC2209 driver through the MCU. The driver accepts those steps, interpolates additional microsteps between them, and sends 256 actual microsteps to the motor. That way the movement is more fluid and quieter. The drawback is the delay between accepting original steps and executing interpolated ones. This provides a small positional error and thus is not recommended by the official Klipper docs. But the error is proportional to specified microsteps. It may be somewhat significant if you're using interpolation with 16 microsteps (~0.006 mm) in some applications, but negligible with 64 (~0.0015 mm) on 99.9% of home-use printers

andrewmcgr, 2025-01-02:
Exactly this.

On top of which, Klipper doesn't synchronise the extruder very precisely,
and that issue is way more important than interpolation.

closed 2025-06-25 by mjonuschat