andrewmcgr/klipper_tmc_autotune #212: Autotune introducing severe whine while holding for LDO-35STH28-1004A

eprouty, 2024-11-02:
```
[motor_constants LDO-35STH28-1004A]
# Coil resistance, Ohms
resistance: 3.2
# Coil inductance, Henries
inductance: 0.0035
# Holding torque, Nm
holding_torque: 0.11
# Nominal rated current, Amps
max_current: 1
# Steps per revolution (1.8deg motors use 200, 0.9deg motors use 400)
steps_per_revolution: 200
```

Hey folks! Just defined this motor based on the data sheet provided by DFH.
[image]

When I enable this it produces a very audible whine while holding, something that is not present without autotune enabled for this one motor.  This is the only motor that this is happening on.  The other 7 motors between my Trident and ERCF v2 I have it defined on do not experience this.

Did I do something wrong defining the motor? Is there something else I am missing?  Any help is appreciated, thank you!

LabunskyA, 2025-01-03:
Not sure about the exact cause (I do experience this as well), but this should not be indicative of any configuration problems. You can always just disable motors after homing for idling

mjonuschat, 2025-07-31:
Can you share your klipper and autotune config for the motor/stepper as well (e.g. tuning goal)

closed 2025-09-07 by mjonuschat