andrewmcgr/klipper_tmc_autotune #217: Instantly Blown TMC5160 Chip

mbrady4597, 2024-11-25:
After configuring using autotune as outlined in the guide, one of my TMC5160 drivers overheated and popped on first movement. I have FYSETC TMC5160 QHV chips running at 50V, on a BTT Octopus Pro V1.0 F429 board and powering LDO 42STH48-2804AH motors. The place the chip blew seems to be a common location when used with high voltage. Searches elsewhere seem to indicate that the breakout boards used with the chips do not adequately protect the onboard voltage regulator under all circumstances. https://www.reddit.com/r/VORONDesign/comments/uakvt8/there_are_reports_stepper_drivers_popping_they/ 

Referring to the tmc5160 datasheet, the voltage regulator must reduce the supplied voltage down to 11.5V and overheating can be exacerbated by high PWM freq, high clock freq, and high supply voltage. I believe the settings that autotune produced caused the malfunction. A notable difference from my previous settings was pwm_freq = 2 instead of 0. I can't say this was the cause for certain, but this would put the freq toward the edge of the recommended range from the datasheet.

My previous driver settings (using 64 microsteps and same for x and y):

[tmc5160 stepper_x]
cs_pin: x_uart_pin
spi_software_mosi_pin: stepper_spi_mosi_pin
spi_software_miso_pin: stepper_spi_miso_pin
spi_software_sclk_pin: stepper_spi_sclk_pin
stealthchop_threshold: 0
interpolate: True
run_current: 2.0 #max 2.8
driver_TBL: 0
driver_TOFF: 4
driver_HEND: 0
driver_HSTRT: 4

This is the report that came up in the klippy.log: 

TMC 'stepper_y' reports DRV_STATUS: ffffffff sg_result=1023 s2vsa=1(ShortToSupply_A!) s2vsb=1(ShortToSupply_B!) stealth=1 fsactive=1 cs_actual=31 stallguard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
Transition to shutdown state: TMC 'stepper_y' reports error: DRV_STATUS: ffffffff sg_result=1023 s2vsa=1(ShortToSupply_A!) s2vsb=1(ShortToSupply_B!) stealth=1 fsactive=1 cs_actual=31 stallguard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1

The settings also caused undervoltage reports on all steppers (Z steppers are LDO 42STH48-2504AC):

TMC 'stepper_z' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_z' reports GSTAT:      00000000
TMC 'stepper_z1' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_z1' reports GSTAT:      00000000
TMC 'stepper_z2' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_z2' reports GSTAT:      00000000
TMC 'stepper_x' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_x' reports GSTAT:      00000000
TMC 'stepper_x1' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_y' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_x1' reports GSTAT:      00000000
TMC 'stepper_y' reports GSTAT:      00000000
TMC 'stepper_y1' reports GSTAT:      00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
TMC 'stepper_y1' reports GSTAT:      00000000

The following are the full settings from klippy.log:

autotune_tmc set stepper_x pwm_freq=2
autotune_tmc stepper_x ncycles=219 pfdcycles=73
autotune_tmc set stepper_x tpfd=1
autotune_tmc set stepper_x tbl=1
autotune_tmc set stepper_x toff=1
autotune_tmc seting hysteresis based on 50.0 V
dcoilblank = 0.166667, dcoilsd = 0.019661
hysteresis = 33, htotal = 14, hstrt = 8, hend = 6
autotune_tmc set stepper_x hstrt=7
autotune_tmc set stepper_x hend=9
autotune_tmc using max PWM speed 440.177100
autotune_tmc set stepper_x pwm_autoscale=True
autotune_tmc set stepper_x pwm_autograd=True
autotune_tmc set stepper_x pwm_grad=4
autotune_tmc set stepper_x pwm_ofs=13
autotune_tmc set stepper_x pwm_reg=15
autotune_tmc set stepper_x pwm_lim=4
autotune_tmc set stepper_x tpwmthrs=1048575
autotune_tmc set stepper_x en_pwm_mode=False
autotune_tmc set stepper_x tcoolthrs=313(30.0)
autotune_tmc set stepper_x sgt=1
autotune_tmc set stepper_x small_hysteresis=False
autotune_tmc set stepper_x semin=2
autotune_tmc set stepper_x semax=4
autotune_tmc set stepper_x seup=3
autotune_tmc set stepper_x sedn=2
autotune_tmc set stepper_x seimin=1
[... 235 lines cut ...]
New driver settings (same for x and y):

[tmc5160 stepper_x]
cs_pin: x_uart_pin
spi_software_mosi_pin: stepper_spi_mosi_pin
spi_software_miso_pin: stepper_spi_miso_pin
spi_software_sclk_pin: stepper_spi_sclk_pin
stealthchop_threshold: 0
interpolate: True
run_current: 2.0 #max 2.8
driver_TBL: 1
driver_TOFF: 1
driver_HEND: 9
driver_HSTRT: 7
driver_TPFD: 1
driver_PWM_FREQ: 0
driver_PWM_GRAD: 4
driver_PWM_OFS: 13
driver_PWM_REG: 15
driver_PWM_LIM: 4
driver_SEMIN: 2
driver_SEUP: 3
driver_SEMAX: 4
driver_SEDN: 2
driver_SEIMIN: 1

mjonuschat, 2025-06-25:
Hi @mbrady4597!

I’m closing this issue because there haven’t been any more reports of TMC autotune causing TMC5160 failures, except for the one you’ve reported. It seems like this is just a coincidence, not a bug in TMC autotune. The same failure mode has been reported on Reddit and Discord servers by many users who don’t use TMC autotune.

closed 2025-06-25 by mjonuschat

kaibsora, 2025-11-13:
I would like to reopen this as still an issue. using the almost identical setup as them, my x1 driver blew in the exact same place. 
Same drivers(just have 8 of them)
octopus pro f446(instead of his octopus f429)
50v input from meanwell psu(lrs-600-48)
latest klipper/tmc autotune.
8x ldo-42sht48-2504ah

After replacing and lowering my 48v voltage from 50 to 48, i still cannot run tmc autotune at all with the setup. all of the drivers simultaneously report overtemp, despite without autotune included, the system runs fine, and the drivers(which have 3 5020 blowers on them at full blast) are cool as a cucumber. all of my stepper motors also have 5020 axial fans blowing on them as well. My klippy log was also almost identical. I've also tried without specifying the voltage in the conf, but no difference. I'm more than happy to troubleshoot as needed.