andrewmcgr/klipper_tmc_autotune #225: Lower chopper frequency for TMC2240 may not be required

liampwll, 2024-12-27:
This recent commit changed the target PWM frequency of the TMC2240 to 20kHz to reduce the power dissipation: https://github.com/andrewmcgr/klipper_tmc_autotune/commit/fd68808fac45c367cdc22d01bce27b4e3be81ccc

I suspect that any increased power dissipation compared to the TMC2209 is due to the output slew control that defaults to 100V/us but can be increased up to 800V/us, giving us a faster slew rate than the 2209.

If we assume a motor running at 2A and 24V the dissipation can be reduced by an entire watt:
```
55kHz×2A×(24V/2)×(24V/(100V/us))×4 = 1.2672W
55kHz×2A×(24V/2)×(24V/(200V/us))×4 = 0.6336W
55kHz×2A×(24V/2)×(24V/(400V/us))×4 = 0.3168W
55kHz×2A×(24V/2)×(24V/(800V/us))×4 = 0.1584W
```

mjonuschat, 2025-06-24:
I think this has been addressed with a6c2658?

liampwll, 2025-06-24:
PWM_FREQ_TARGETS has not been adjusted to match the others however they should no longer run hotter when using this.

mjonuschat, 2025-08-08:
> PWM_FREQ_TARGETS has not been adjusted to match the others however they should no longer run hotter when using this.

Ran a few experiments and with 55kHz the 2240s still run 10-15C hotter than at 20kHz. Given that the frequency is configurable I'd say the lower chopper frequency is a good default.

closed 2025-08-08 by mjonuschat