andrewmcgr/klipper_tmc_autotune #228: Request to add: MS17HD6P4201-04

Sonnenwerk, 2025-01-02:
[motor_constants MS17HD6P4201-04]
coils resistance: 1.3
coil inductance: 0.0027 (2.7 mH) 
holding_torque: 0.67NM
max_current: 2.0 A
steps_per_revolution: 200
Rated Voltage 2.6V
Insulation class B 130 degree

# R is coil resistance, Ohms
# L is coil inductance, Henries
# T is holding torque, Nm (be careful about units here)
# I is nominal rated current, Amps

dvystrcil, 2025-05-20:
I think you have the model incorrect. The final 1 is an I. 

[image]

Notice that the MS1... has a serif, whereas the ...0I-04 does not.

You will also find the MS17HD6P420I-04 in the database:
https://github.com/andrewmcgr/klipper_tmc_autotune/blob/main/motor_database.cfg#L274

mjonuschat, 2025-06-23:
Closing as the motor is already in the DB as `moons-ms17hd6p420I-04`

closed 2025-06-23 by mjonuschat