andrewmcgr/klipper_tmc_autotune #233: Request to add: BJ42D15-26V19

Linearlink, 2025-01-17:
[motor_constraints BJ42D15-26V19]
resistance: 6.0
inductance: 0.0088
holding_torque: 0.28
max_current: 0.84
steps_per_revolution: 200

clarin-ebtio800090, 2025-01-17:
Hello, we tried to solve the issue.

This is what we did:

Added the specifications for the BJ42D15-26V19 motor to the motor_database.cfg file. This includes resistance, inductance, holding torque, max current, and steps per revolution as provided in the criteria.

You can review changes in this commit: https://github.com/clarin-ebtio800090/andrewmcgr-klipper_tmc_autotune-233/commit/f2f819f41110f2061ae1d80628bddd65323f14bc.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

stablestud, 2025-01-23:
Hi @Linearlink,

my changes includes your motor. Once my PR has been accepted it should be available for all.

mjonuschat, 2025-06-23:
Close through PR #234

closed 2025-06-23 by mjonuschat