andrewmcgr/klipper_tmc_autotune #241: Request motor Add : SL42STH40-1684A

thedizze, 2025-02-16:
can you gently add SL42STH40-1684A used on most tronxy printers and used for some VZ build? 

thank you

locki-cz, 2025-02-16:
> can you gently add SL42STH40-1684A used on most tronxy printers and used for some VZ build?
> 
> thank you

and can you gently make proper change request with all info what is need?

thedizze, 2025-02-17:
Sorry I'm having difficulties in finding specs about this motors, tronxy doesn't have datasheet avaiable

locki-cz, 2025-02-17:
then it is impossible add it to autotune when you dont know specs :(

mjonuschat, 2025-06-23:
Closing since adding motors without available specs isn't possible. 
If you find the needed information/datasheets in the future feel free to open a new issue/PR

closed 2025-06-23 by mjonuschat