andrewmcgr/klipper_tmc_autotune #252: Bad results with silent profile, on Siboor.

Barafu, 2025-05-24:
Tried this on my Voron on X/Y drives (I am seeking to reduce VFA). In performance mode, it works fine, but I did not notice any improvements either, compared to the voron default config. In silent mode, however, the motors emit screeching sounds when moving, and the head stops in a wrong position, which immediately creates layer shifts. 

I use Octopus Pro, TMC2240, motors are siboor-42sth48-2504. I have checked that the motors data in DB equals the data published by the developer. 

Here is the relevant part of my config: 
```
#####################################################################
##                   X-axis on MOTOR_0(B Motor)
#####################################################################

[stepper_x]
step_pin: PF13                           # X-axis motor pulse pin setting
dir_pin: PF12                           # X-axis motor direction pin setting
enable_pin: !PF14                        # X-axis motor enable pin setting
endstop_pin: EBBCan:gpio24               #PG6 # Limit switch pin setting (X-)
rotation_distance: 40                    # Active pulley circumference in mm (2GT-20T pulley 40, 2GT-16T pulley 32)
full_steps_per_rotation: 200             # Number of pulses required for a single motor revolution (1.8 degree motor: 200, 0.9 degree motor: 400)
microsteps: 128
position_min: 0                          # X-axis minimum travel - software limit
position_endstop: 350                    # Mechanical reset point coordinates for X-axis (change to 350 for 350 models)
position_max: 350                        # X-axis maximum travel - software limit (change to 350 for 350 models)
homing_speed: 100                        # Reset speed - maximum 100
homing_retract_dist: 5                   # Setback distance after the first trigger of the reset switch

[tmc2240 stepper_x]
cs_pin: PC4                              # Chip select pin for TMC2240 driver
spi_software_sclk_pin: PA5               # SPI software clock pin
spi_software_mosi_pin: PA7               # SPI software MOSI pin
spi_software_miso_pin: PA6               # SPI software MISO pin
run_current: 2                        # Motor running current value (A)
interpolate: true

[autotune_tmc stepper_x]
motor: siboor-42sth48-2504(s45)
tuning_goal: performance
```

Both motors and drivers sit around 70C

locki-cz, 2025-05-24:
I think you should reduce run_current: 2  to something like 1.1 to 1.4 ... before also use [github.com/Frix-x/klippain-shaketune](github.com/Frix-x/klippain-shaketune) and tune and find speeds with low vibrations from your motors, you can also define voltage on motors, tmc2240 can be used also on 36V, not know if it is your case ...

mjonuschat, 2025-07-31:
I’m closing this old issue. If it’s still happening, please open a new one and add information regarding speeds used in silent mode.

closed 2025-07-31 by mjonuschat