andrewmcgr/klipper_tmc_autotune #277: How does autoswitch work?

Jademalo, 2025-07-29:
I know it's not advisable due to step loss during the switch, but I've been curious to test it regardless.
After looking through the code, I'm a little puzzled as to how it actually decides on the crossover point from stealthChop to spreadCycle.

I can see that the mode [enables stealthChop](https://github.com/andrewmcgr/klipper_tmc_autotune/blob/09d85f352a4c83c1376a6c2813ebab7d6fd2053e/autotune_tmc.py#L349), and I'm assuming that `tpwmthrs` is the crossover value between the two, being set to `0` for `silent` and `0xfffff` for `performance`. What I don't understand is the difference between `_set_driver_field` and `_set_driver_velocity_field`.
I'm assuming that it gets `pwmthrs` from the motor constants as the max pwm speed it can handle, and then passes that down through `_set_driver_velocity_field `, but I can't see how it actually works out the crossover point.

Is there any way to enable logging to see where exactly it's setting the crossover?

Thanks

mjonuschat, 2025-07-31:
`_set_driver_field` directly sets the chip field to the value. `_set_driver_velocity_field` calculates the correct value for the field first, based on the requested switching velocity and configuration of the stepper driver (e.g. effective step distance) before then programming the driver field. The actual calculation happen in Klippers `TMCtstepHelper`.

As to the switching point, that should show up in your klipper.log with a message similar to this:

```
autotune_tmc set stepper_x tpwmthrs=1048575
autotune_tmc set stepper_y tpwmthrs=1048575
```

Jademalo, 2025-08-01:
Thanks, appreciate the answer!
I've not been getting anything in klipper.log for some reason, I did go looking but there wasn't anything.

Is it possible to define the requested switching velocity anywhere? I've got a really particular Z axis that really doesn't like higher speeds on stealthChop, but it would be really useful to be able to disable it and allow for faster travel for the big z move when homing after a tall print.

Jademalo, 2025-08-09:
Ah alright, thanks for checking!
In that case It might be useful to have a debug output to show where the crossover speed is, for slicer profile tuning purposes.

commit 8874d2ea (2025-09-07): Feat: Log StealthChop autoswitch velocity limit

Log the velocity limit where switching between StealthChop and SpreadCycle mode will happen when the tuning goal is set to `autoswitch`.

Closes #277