andrewmcgr/klipper_tmc_autotune #288: Install failed

jmorg1027-droid, 2025-09-13:
Trying to install on FLSUN V400 SpeederPad however keep getting error message "[ERROR] Klipper service not found, please install Klipper first!". I've attempted to copy the path directly from the main readme page and also from the raw page, close terminal and restarted system several times still same error message. Also attempted sudo  apt-get update --allow-releaseinfo-change and sudo apt full-upgrade -y commands, still same error and receive the same error when sudo service klipper start and systemctl start klipper commands. Also, I am aware of at least one person using it on a FLSUN V400 because they have a printer.cfg file with it listed in it, that's where I originally learned about it. Here is a link to his page with the printer.cfg files:  https://github.com/Aieie78/FLSUN_V400_Config/blob/main/printer.cfg
Attached is a screenshot of my terminal, hopefully it loads. Please let me know if any other information is needed, thank you looking forward to using the tmc_autotune system.

[image]

LabunskyA, 2025-09-15:
Can you verify that your printer and all of the software are ready-to-print at the time of autotune installation?

jmorg1027-droid, 2025-09-15:
Yes, I've been printing with it. Had someone ask if I had installed official Kilpper, and that may be the issue, it could still be on FLSUN factory Klipper. Have to figure out someway of checking to see if I've installed official Klipper or not, I can't remember.

mjonuschat, 2025-09-16:
For troubleshooting that error the output of `sudo systemctl list-units --full -all -t service --no-legend` (run over SSH on the printer) would be helpful.

jmorg1027-droid, 2025-09-19:
Thank you very much. Due to emergency health issue of my fiance I haven't been able to fool with my printer in the last couple of days but I imagine its because of me running multiple klipper instances from the same SpeederPad. I will definitely try this solution to change the directory and see if that works as soon as I get back around to fooling with them. Again I will definitely update once I figure out which solution works. Thanks again guys!!

jmorg1027-droid, 2025-09-20:
Update.... Thanks again everyone for all of your help. It was due to running multiple instances of Klipper on the same SpeederPad. Followed @2904nando suggestions and created a fork and edited L22 and L75 with the appropriate service address, mine happened to just be labeled Klipper-1. Its up and working properly now. Have to do some test prints and try out the tmc_autotune features. Thanks again everyone, hope this issue helps everyone in the future.

closed 2025-09-20 by jmorg1027-droid